CTCaer/hekate #515: Cannot Read Game Cards

KidLycan, 2020-12-03:
What's the issue you encountered?

When finishing the normal launch after doing the copy paste update method for the 11.0.0 switch update, after booting up as CFW, an error message stating that the game card couldn't be read. tried with Mario cart 8 Dlx and them Pokemon sword to see if the game cartage itself was the issue. It wasnt. I booted it to the Stock in the launch menu were CFW is and after it booted, the cartage was being properly read and able to be played!
How can the issue be reproduced?

After copy-paste update, try booting up and launching CFW
once you go to the main switch start up screen where you see the top 3 newest news, before a input is sent, a cannot read came card message pops up. And it happens after you insert a different and the same cartage. (and did the blow into the cartage gimmick too XD)
System Firmware Version

Switch: 11.0.0
Hekate: 5.5.0 https://github.com/CTCaer/Hekate/releases/
Nyx: v0.9.6 https://github.com/CTCaer/Hekate/releases/
Atmosphère: 0.16.0 (https://github.com/Atmosphere-NX/Atmosphere/releases)
Additional context?

I have one of the very older few models that can use the 10 pin way to get into the RCm mode.

All and all, its still able to play games cartage by going through the Stock launcher and able to play downloaded games from the CFW launcher. I just wanted ya to know about the issue.

(reposted to make sure both can see if this error is one or both causing the issue.) 
https://github.com/Atmosphere-NX/Atmosphere/issues/1237

isabellajj, 2020-12-04:
I have the same issue. 
And I have done the changes of nogc=0 in BCT.ini as per SciresM advice

urherenow, 2020-12-04:
Are you loading from hekate or fusee-primary?

blackmanjones3, 2020-12-05:
Yeah i did the nogc = 0 and my switch and i still have the same problem and im loading from hekate

urherenow, 2020-12-05:
loading from hekate does not use BCT.ini, please look at https://github.com/CTCaer/hekate/blob/master/res/hekate_ipl_template.ini

autonogc=0 would be the equivalent...

blackmanjones3, 2020-12-05:
@urherenow dude you are a life saver thank you

isabellajj, 2020-12-05:
Same, thank you! It finally works now. thank you

CTCaer, 2020-12-05:
You don't need to edit hekate_ipl.ini by hand btw.
Like most relevant settings, you can just go into options and change the AutoNOGC in there.

closed 2020-12-05 by CTCaer

anaxxo, 2021-02-06:
I'm getting the same error, but using Fusee Primary , is there a way to fix it from options like in hekate ? @CTCaer 

Thanks in advance

urherenow, 2021-02-06:
THIS issue is closed, and this isn't supposed to be a help forum. Fusee-primary is vanilla Atmosphere, so your fix is in atmosphere's BCT.ini. nogc=0.