CTCaer/hekate #518: updated to hekate 5.5, how do I download atmosphere 0.17.0. ?

wetzthebetz81, 2020-12-04:
Where can I download Atmosphere 0.17.0  ?

Roxas-Dev, 2020-12-04:
Atmos 0.17.0 hasn't been released yet 

https://github.com/Atmosphere-NX/Atmosphere/releases

wetzthebetz81, 2020-12-04:
Yes I know. So why does the release of hekate 5.5. say Atmosphere has to be updated to 0.17.0?

It is not compatible yet.

masagrator, 2020-12-04:
Because it was announced that 0.17.0 will have Mariko support, so futureproofing

wetzthebetz81, 2020-12-04:
Ok, that makes sense. Thanks. I thought that was my problem because my games still ain't working. I updated my Switch to 11, updated atmosphere to 0.16.0 and updated hekate tot 5.5. Am I missing something?

urherenow, 2020-12-04:
Depends when you updated AMS. The build was re-uploaded with bugfixes.

wetzthebetz81, 2020-12-04:
Thanks. Could it be the Sigpatches? I didn't update them.

urherenow, 2020-12-04:
If you're trying to play pirated software, then yes, it could be. If carts aren't working you need to disable nogc and reboot, keeping in mind that downgrading might not work very well after upgrading your gamecard slot. IMHO, it's not worth it to downgrade anyway.

CTCaer, 2020-12-05:
Mariko supported atmosphere (0.17.0) will be released when it will be ready.

Also no piracy mentions

closed 2020-12-05 by CTCaer