CTCaer/hekate #520: hekate_ctcaer_5.5.0.bin gives me an error when I try to inject it to my Nintendo Switch.

LuisDennis, 2020-12-05:
After updating hekate to v5.5.0 I get an error that says:
`Failed to mount SD!`
`Missing LP0 (sleep mode) lib!`
`Missing old Minerva lib!`

`Update bootloader folder!`

`Press any key...`

I already had the bootloader from v5.5.0 on my microSD Card. Injecting an older version of hekate like hekate_ctcaer_5.3.4.bin to my Nintendo Switch (even with the v5.5.0 bootloader folder on my microSD Card) works and using a different microSD Card with the hekate v5.5.0 bootloader folder works. I'm using this microSD Card [https://i5.walmartimages.com/asr/5ad3ec8d-a9a4-4406-927b-e88054fb2ba0_1.f2a8686ba7492332a91cd6ff3a07c801.jpeg].

CTCaer, 2020-12-05:
What sd card is that

Get a screenshot or photo from `Console info` -> `Sd card`.

LuisDennis, 2020-12-06:
[image]

urherenow, 2020-12-06:
I wonder if when you right-click on the zip file and go to properties, if there is an "unblock" checkbox there, or if for some strange reason the archive bit is set on your bootloader folder.

LuisDennis, 2020-12-06:
> I wonder if when you right-click on the zip file and go to properties, if there is an "unblock" checkbox there, or if for some strange reason the archive bit is set on your bootloader folder.

It was blocked, after unblocking it I retried with the files from the unblocked .zip and it still gave me the same error.

CTCaer, 2020-12-06:
@LuisDennis dm me via retronx discord when you have time for testing.

LuisDennis, 2020-12-07:
> @LuisDennis dm me via retronx discord when you have time for testing.

What's your Discord username? Mine is Loser Monkey#5556.

CTCaer, 2020-12-07:
Try to force shut off (Press power for 12s). And then inject hekate. Tell me what happens.

samwelnella, 2020-12-09:
I am having a similar issue. Using hekate 5.5.0 payload I get this error:
`Failed to mount SD!
Missing LP0 (sleep mode) lib!
Missing old Minerva lib!

Update bootloader folder!

Press any key... `

Using hekate 5.3.4 payload my SD card works fine.

[This](https://www.bestbuy.com/site/pny-pro-elite-512gb-microsdxc-uhs-i-memory-card/6329148.p?skuId=6329148) is the microSD card I’m using. And the SD card info is as follows:

[image]

CTCaer, 2020-12-09:
Join retronx discord and pm me. This can't be easily debugged here.

CTCaer, 2020-12-10:
So @samwelnella helped me debug this and the issue is weird.
It seems to only happen on these PNY (Phison made) sd cards, based on the current 2 reports.

What causes it was identified, but needs investigation on the why.
Because it was something added that was making the driver more close to HOS one.

For now here are the 2 binaries that mitigate it:
[mitigation.zip](https://github.com/CTCaer/hekate/files/5669984/mitigation.zip)
nyx.bin goes into /bootloader/sys/.

@LuisDennis Please reply if that works for you also.

acedaemon, 2020-12-11:
I get the same error message with 5.5.0 
The mitigation works except when using reboot to payload from the homebrew menu
And I have a SanDisk
[image]

CTCaer, 2020-12-11:
@acedaemon You are probably describing another issue.
Does it work if you just reboot from the power menu?

acedaemon, 2020-12-11:
Yes, that reboots to hekate. Do you want me to open another issue then?

CTCaer, 2020-12-11:
@acedaemon That's an issue with the bundled r2p homebrew in atmosphere.

acedaemon, 2020-12-11:
@CTCaer Alright. Thanks for the mitigation

CTCaer, 2020-12-11:
@acedaemon 
You are welcome but the mitigation doesn't affect you.
The issue described by LuisDennis and samwelnella only happen on these Phison cards.

Your case can be fixed on a new atmosphere release. Otherwise you can recompile it yourself and changing 
https://github.com/Atmosphere-NX/Atmosphere/blob/master/troposphere/reboot_to_payload/source/main.c#L7
from `0x20000` to `0x24000`

LuisDennis, 2020-12-13:
hekate v5.5.1 fixed the error.

CTCaer, 2021-01-05:
The issue was extensively debugged but there's no solution.
Seems like an issue with the card's first init. hekate will continue to mitigate that.

closed 2021-01-05 by CTCaer