CTCaer/hekate #521: Can’t boot into lineageOS

FatalPunisher88, 2020-12-06:
Can’t boot into LineageOS anymore since update. 
Firmware 11.0.0/AMS 16.0
Hekate 5.5.0

Screen just goes black and I have to restart my switch.

JteMonteEnDerby, 2020-12-06:
it does me the same but me i can't even start Athmosphere

JteMonteEnDerby, 2020-12-06:
The new maj broke everything

urherenow, 2020-12-06:
First thing to try (for Atmosphere problems): rename/remove atmosphere/contents and replace it from the release zip.

If it works, your issue is with a theme or something with a boot2.flag in it (like no-ovlloader).

CTCaer, 2020-12-06:
@JteMonteEnDerby as always with every major HOS version and not, you need to clean your content folders. Removing themes, removing incompatible with HOS sysmodules and incompatible with new Atmosphere sysmodules.
So as always the first check/test is to rename Atmosphere folder and paste a fresh one.

@FatalPunisher88 which lineageos android version?

JteMonteEnDerby, 2020-12-06:
@CTCaer i have try with only atmosphere file and it don’t work too

FatalPunisher88, 2020-12-06:
@CTCaer it is 
LineageOS 15.1 - Android 8.1 Oreo

I usually boot Atmosphere,Sxos and Android from Hekate and it has all been working fantastic until I updated.i am just checking again now and it seems SX OS 3.10 and LineagaeOS give me the black screen when booting from Hekate.and I have to restart the switch.atmosphere boots fine from Hekate[image]

JteMonteEnDerby, 2020-12-06:
Yea yea there is definitely a thing because I ask to a YouTuber who show smash mod if he got problem and he say yes

CTCaer, 2020-12-07:
Android 8.1 is not supported. It uses an ancient and broken uboot/kernel combo (1.5 years old).
And boot is hit or miss on it.
Supporting it with **active mitigations** is not something that I'm willing to do.
Especially when the android devs had more than a year to do a single and simple coreboot.rom update to make it robust.

As I stated in switchroot discord, Android 8.1 must die.
You should update to Android Q.

If you want further comments or help about it, @bylaws can help you.

closed 2020-12-07 by CTCaer