CTCaer/hekate #524: Comm failure 3 while using USB Tools

mattlev, 2020-12-08:
I've tested a couple cables, USB ports, and files, they work for transferring the same file over the same cable on the same port to my phone.  The SD Card is formatted FAT32 and the file size is under 4GB, I can transfer it to the SD card if I put the SD card in an SD card reader and put it into my computer.  When using USB Tools the switch SD card is recognized, I can browse it, but if I try to make any changes USB tools tells me there's a comm failure, windows loses connection to the device, and USB tools says it was ejected.
I'm using rekado to inject hekate from RCM

urherenow, 2020-12-08:
It might be that I'm on mobile and don't see the whole issue, but I don't see what version you're using or whether you installed the usb drivers on your pc or not.

mattlev, 2020-12-08:
Sorry, version 5.5.0, and I've installed drivers on PC (tried through Tegra and NS_USBLoader)?

mattlev, 2020-12-08:
Turns out the switch just didn't like all three cables I initially tried to use.  a fourth cable I grabbed just magically worked.  super weird.

closed 2020-12-08 by mattlev

urherenow, 2020-12-08:
If you have a meter handy, check for continuity from the outside of the connectors between the ends of the cable (see if the cables have end-to-end grounded shielding). In all cases where I have run into a problem like this with micro USB cables, there was none...