CTCaer/hekate #526: Latest Hekate - Can't run Android payload

jlaidlaw86, 2020-12-09:
I am on the latest version of Hekate. When I try to launch android payload I get an error message telling me that coreboot is disabled . I am then returned to Hekate home screen. 

my switch is mariko with sx core. I run hekate as a payload from boot.dat

What do i need to do to get android running?

CTCaer, 2020-12-09:
Read the changelog.
coreboot.rom which is used for android and linux is forbidden on mariko.

closed 2020-12-09 by CTCaer

SteFlyn, 2020-12-10:
I have Erista V1 and I'm also greeted by a blackscreen when I try to boot android. Downgraded Hekate and it works now.

CTCaer, 2020-12-10:
@SteFlyn Read my response on https://github.com/CTCaer/hekate/issues/521
Asks the devs for updated coreboot/dtb if you still want to run that broken version.