CTCaer/hekate #527: Bit of a long shot but here goes

psxpetey, 2020-12-10:
Switch has a blue screen biskeys works, and most text based payloads.

Hekate will send and load but is a black screen. With sd card removed hekate will load the text version.

Fuses etc. When I check Emmc it gives partition info gpp partition table has no info under it?

If I insert the as card text based hekate can not mount it. If I insert it beforehand even with no boot loader file it freezes/ blackscreen as soon as I remove the sd the text version comes right up. As card seems to work ok in computer.

Fusee tmosphere fails: fatal exception in text

What can I do about gpp?
Why won’t hekate accept sd card when biskeys uses it just fine

CTCaer, 2020-12-10:
Hekate relies on SD that's why.

There's something seriously wrong with your switch. Based on your reaction I would guess you got it second hand?

Seems like an issue with SD reader and your emmc seems wiped.

Try to boot hekate with SD. Remove the Minerva .bso file from bootloader/sys first and then wait on it after injection.

If it's a pin issue there's a chance it will load after a minute or so.

Thisisntmeasfarasiknow, 2020-12-12:
This is not a software/Hekate issue. 

It's by complete chance i stumbled on this....i was looking to solve an unrelated issue.

This is a hardware issue

The guy was given very clear and thorough instructions to diagnose and repair a hardware related fault on a forum post he made, unfortunately he's omitted most of this info to you and was unable to follow directions for repair, was also "checking for shorts on the gpu" with the battery connected...........

originally it was a RAM issue and was advised to reflow to resolve glitchy graphics.... which solved that issue but he was unable to follow simple diagnosis and repair steps which were clearly laid out for him, wants please-bro solutions

As it stands it's a suspected SOC issue, not EMMC, SD or hekate.

The board was likely considered a donor...time waste.

psxpetey, 2020-12-12:
> This is not a software/Hekate issue.
>
> It's by complete chance i stumbled on this....i was looking to solve an unrelated issue.
>
> This is a hardware issue
>
> The guy was given very clear and thorough [...]

imagine being such a huge loser you are now chasing my posts around the internet shgut the fuck up dom. 
anyway it was found finally to be a software issue with the emmc. get a life dude.

psxpetey, 2020-12-12:
> Hekate relies on SD that's why.
>
> There's something seriously wrong with your switch. Based on your reaction I would guess you got it second hand?
>
> Seems like an issue with SD reader and your emmc [...]

thanks it was found to be an issue with the emmc, the person who owned it before me messed it up or deleted it.

closed 2020-12-12 by psxpetey