CTCaer/hekate #528: T210b01: coreboot not allowed!

WTZ91, 2020-12-10:
I had upgraded to OFW 11. 

I found hekate 5.5 and l4t ubuntu. After installation, when I try to load ubuntu got this error.  

Does anyone have similar issues?

balika011, 2020-12-10:
Coreboot doesn't work on mariko.

WTZ91, 2020-12-10:
Will it be supported in the future?

CTCaer, 2020-12-10:
Already answered and also it's stated in changelog.
coreboot.rom is disallowed on Mariko for your own protection.
Support will be ready when it's ready basically.

closed 2020-12-10 by CTCaer

ShadePhenix, 2021-07-11:
@CTCaer 
Hi!

First of all, thanks for the great work you are doing!

I have searched but couldn't find the answer to my question.
As far as I understand, coreboot.rom is still not allowed on Mariko for protective reasons, and there is no timeframe for when this will be supported.
My question is:
Is there any other way for booting into Ubuntu/Android? Or is the only option to wait for support?

Many thanks.

SP

lmm1191, 2021-11-22:
Is there a progress on running Lakka/Linux/Android for Mariko?

greymanwalking, 2021-12-12:
I created an account just to ask when will this be supported on switch mariko device ?
Would really like Android/Ubuntu on my switch pleaseee

CTCaer, 2021-12-12:
Android/Ubuntu is not supported on mariko.
If it ever gets a support you will hear it.
(Also coreboot will not be used there)

Masamune3210, 2022-05-12:
You did. Believe me there will be surely lots of noise if Mariko ever gets support. If you wanted a tablet strictly for Android, why didnt you just buy a android tablet?

krzysztofkrzeslak, 2022-05-13:
my objective is to have Ubuntu or some other plain Linux distro, not android so there's not much choice then and Switch seemed like the best choice, which now seems wasn't, so I guess I need to sell and look for a V1 version.

CTCaer, 2022-05-14:
L4T is not ported to mariko (v2/lite/oled) yet