CTCaer/hekate #531: Booting into CFWsysNAND boots into CFWemuMMC

HearMeWhisper, 2020-12-12:
When I expect to be booted up into my sysNAND with CFW, settings indicate I am booted into emuMMC instead, even when CFW sysNand was chosen.  I'm not sure what kind of data you need to debug this, but happy to provide whatever I can.

urherenow, 2020-12-12:
hekate_ipl.ini

CTCaer, 2020-12-12:
If your boot entry does not define anything extra other than fss0, it will adhere to emummc.ini.
So if emuMMC is enabled it will boot that. If not, it will boot sys.

For that reason hekate supports 2 special keys in order to protect user and make sure what is booted is the wanted one.

These 2 are:
`emummc_force_disable=1`
- Disables emuMMC for that boot entry even if it's globally enabled.
- Ensures that the selected boot entry will boot sys.

`emummcforce=1`
- Ensures that emuMMC is enabled and bootable.
- Errors out if it's not, protecting the user from running sys.

So 2 super simple boot entries can look like that:
```ini
[Atmo SYS]
fss0=atmosphere/fusee-secondary.bin
emummc_force_disable=1

[Atmo EMU]
fss0=atmosphere/fusee-secondary.bin
emummcforce=1
```

First one ensures that SYS will be booted.
Second one depends on a valid emuMMC, otherwise it errors out as explained above.

If you want to learn more about the ini and boot entry keys supported, read the readme.

closed 2020-12-12 by CTCaer