CTCaer/hekate #532: Android partition swallows Linux

Biyoni, 2020-12-14:
When partitioning my SD card, no matter the values I set, if I enable android and linux partitions together, android tends to swallow up the Linux partition, making one big android partition and no Linux.

To reproduce, use the partition manager, assign any space for Linux, and then assign any space for Android, and android will take up the space for both combined.

CTCaer, 2020-12-14:
Yep, that's correct. That huge partition is a GPT protective partition.
And basically re includes everything that MBR has.

Linux and ofc android, if they see a GPT they completely discard MBR. And thus there's no point to exist in the MBR table.
On the other hand, if android is not enabled, everything can fit in MBR.
In either case, the only partitions that need to exist in MBR are the FAT one and emummc.

In order to see what I mean you need to use linux to view the partition table in the card. Windows completely discards GPT if MBR has more than one partition (which is wrong behavior btw).

Anyway, the partition exists. Just try to flash linux and you will see it will tell you it found a partition.

closed 2020-12-14 by CTCaer

Biyoni, 2020-12-14:
just checking, I'm viewing partitions in partition manager in hekate, not windows, but what you're saying is that one big partition contains it's own GPT which stores the linux and android partitions?

CTCaer, 2020-12-14:
It stores everything. Even fat partition and emummc.
And yes hekate doesn't show it, because it needs lots of GUI bs to show everything.
So I only show the MBR partitions on that pop up.

CTCaer, 2020-12-14:
On the **same** sd:
MBR:
[image]

GPT:
[image]

entry 1 is fat partition, it's the same partition with partition 1 in MBR
entry 2 is the `l4t` partition (linux)
entry 11 is emummc, it's the same partition with partition 2 in MBR
the other entries are android related.