CTCaer/hekate #533: Botting stock after power down emmunand

rubensk8er, 2020-12-14:
Hi im having problem to reboot on stock after power down emmnand, it get stuck on the nintendo switch logo and i have to push the power button to restart and then it boot the stock, im using atmosphere 0.16.1 and made the emmu with the last release of hekate.

 I have a 256 gb micro sd evo select samsung btw.

balika011, 2020-12-14:
Did you check if there is an existing issue for that? I'm pretty sure I read that before.

rubensk8er, 2020-12-15:
Thanks for the response ive only read that is a bug but cant find a solution to that..

CTCaer, 2020-12-15:
I noticed that too when I tested 11.0.0 and 0.16.0 atmosphere on release. And then I entered OFW. Didn't try hekate stock.
I did a small test on 10.2.0 with 0.16.1 and it doesn't do it. So I suppose it's something that was changed in HOS 11.0.0.
I still don't know what causes that.

For now the first OFW boot fails and you need to press pwr once and it reboots automatically.

It's not an issue with hekate, but I will keep that issue open to remind me to research the cause.

CTCaer, 2020-12-15:
I identified the issue and seems to be new regulator checks in 11.0.0+.
What is needed to change is atmosphere's reboot_stub binary that is used to do a power off.

Probably next atmo version will rectify that.

Because of the analysis on how pmic acts on each command, I will also change hekate's power management on reboot/power off also.

closed 2020-12-15 by CTCaer