CTCaer/hekate #534: BSOD (no nintendo logo) + Hekate Black screen (if sd card present), can't restore NAND

zFerry98, 2020-12-14:
For some reason (while using android 9) as usual, the switch rebooted (build bugged). Problem:
Totally blue screen as soon as the console is turned on (no logos / writings etc. only blue screen).
So I tried to start Hekate (5.5.0) as usual... black screen.
If I remove the sd card hekate starts but I can't do anything. If I try to check the info the EMMC gets stuck at GPP.
When Hekate (vertical "console" version) starts I can assure you that the battery has no problems but a bizarre thing. Only the yellow lettering as soon as hekate is injected are clearly seen.
After that, all the time the screen shows strange (uneven) green lines or heavy graphical glitches. It's not the broken screen (it's a weird bug that appears after hekate starts).

Now what I can say is that if I put the sd after starting Hekate, it is seen at 86mb bus 4 (well to say). Why can't I reset NAND in that mode?

I don't remember the firmware version but after verifying that the switch could start the payloads, I no longer started the firmware (it was on 10. something, only now it doesn't start and I can't know what version it was).
I have a full nand backup Boot0/1 + 32gb nand (the one that is done just "cracked" the switch for the first time) but I can't restore it because hekate Full does not start if ANY SD is inserted (I tried 3 different sd) I tried 3 pc (one a laptop) and 2 type-c cables. So I exclude that it is a pc/cable problem.
I have used hekate since 5.3.2 to 5.5.1 and have never had any problems.
If you need screenshots or other info tell me.
Sorry for my English.

Thank you in advance.

CTCaer, 2020-12-14:
Your case interests me a lot. It seems like a hw issue.
I can provide tools for debugging that. Find me in retronx discord.

CTCaer, 2020-12-15:
After further analysis on the issue with the user, the result is hw issue with RAM.
Specifically, based on the graphical artefacts and on that tegra uses channel interleaving on ram (64Bx8) and that it had no single big strips, hints that only one chip is faulty.

User was using switch as media player, so that doesn't hint to any software issue.

Also, it seems that the issues started way before and got worse and worse.
Other components seem to be fine.
That might mean that a simple reflow of the chips might fix the issue. Otherwise reballing or replacement.

closed 2020-12-15 by CTCaer