CTCaer/hekate #536: Unable to launch Hekate

Conando025, 2020-12-20:
So in the past, I used to run kosmos but now switched to pure Atmosphere but in the process, I stopped being able to launch Hekate. If I inject fusse-primary directly I can start Atmosphere, no problem here. If no sd card is present I can inject Hekate 5.5.1 and I get into that pare bones thing that you have to navigate using the Volume buttons. But if my SD is present it says in a yellow font:
```
An exception occurred <LR 810580FC>:
UNDEF

Press any key...
```
Doing so just gets me back to that screen. Any help would be appreciated.

urherenow, 2020-12-20:
Did you actually download and extract the latest hekate AND the LATEST Atmosphere to your SD card?

Conando025, 2020-12-20:
Yeah I checked that 3 times by now

urherenow, 2020-12-20:
If you have hekate_ipl.ini, try removing that temporarily and see if it boots then. Seems strange that you say Atmosphere works, but hekate doesn't. Also seems strange that it sounds like nyx isn't even loading (it should load if nothing but hekate is on your sd).

CTCaer, 2020-12-20:
That error comes from Nyx and it probably means the binary is corrupted.
Make sure you download it again and drag n drop a fresh bootloader folder.

Conando025, 2020-12-20:
So I solved it by moving everything to my pc. Then moved the bootloader files (the same ones as before) and it worked so I moved the rest back. The only thing that stood out to me where some corrupted empty folder in the switch folder but I can't imagine them causing this, so no clue why that happend.

closed 2020-12-20 by Conando025