CTCaer/hekate #537: Sleep Mode Broken Stock (SysNand)

werneck9898, 2020-12-20:
I have a Standard Nintendo Switch (not Mariko or Switch Lite) and after updating to the OFW 11.0.1 on the SysNand and moving to Hekate 5.5.1, when I try to boot into SysNand it works, but sleep mode stays in a black screen and need to reset the system. EmuNand, which is on CFW 10.2/ AMS 14..4 is working fine though, no problems at all. This is my "hekate_ipl.ini":
[image]

Also when I disable Auto RCM and boot directly to the SysNand without Hekate, my sleep mode works alright. Is there any fix without having to disable AutoRCM?

werneck9898, 2020-12-20:
Updating the fusee-secondary.bin to the one from Atmosphere 16.2 fixed the issue for me

CTCaer, 2020-12-20:
Yeah 11.0.0 requires updated atmosphere for sleep to work, among other things.

closed 2020-12-20 by CTCaer