CTCaer/hekate #540: hekate payload bin on release page

ELY3M, 2020-12-23:
I am working on a app for launching payloads.  
I found a way to have payload bins updated on demand via a button.....  
atmosphere github already have fuse-primary.bin for download on their release page.  
https://github.com/Atmosphere-NX/Atmosphere/releases/latest/download/fusee-primary.bin

is it possible that Ctcaer upload hekate_ctcaer.bin to release page?   

I would have to put in unzipping feature and extract exact bin from zip.... 
and put the bin into exact folder.

Masamune3210, 2020-12-23:
you need more than just the payload for hekate to work completely. Nyx has to be kept up to date as well which is what is in that zip file

ELY3M, 2020-12-24:
you are right.....  then i have to do unzipping method and copy bootloader folder.

CTCaer, 2020-12-31:
What @Masamune3210 said. Bootloader folder is crucial and I don't want to promote downloading only the main payload, or have user download 2 things for no reason.

closed 2020-12-31 by CTCaer