CTCaer/hekate #541: Create a default main boot entry

j-james, 2020-12-26:
When setting up a new installation of Hekate, at least one main boot entry needs to be specified in `hekate_ipl.ini`.

It'd be nice if Hekate could detect and append a boot entry to a fresh install.

urherenow, 2020-12-26:
It's left out on purpose (so you don't have to fix your settings with every upgrade).

j-james, 2020-12-26:
Right, but so is the whole file. If you copy over a clean download of Hekate (with no configuration), a new `hekate_ipl.ini` file gets created with the following: 

```toml
[config]
autoboot=0
autoboot_list=0
bootwait=3
backlight=100
autohosoff=0
autonogc=1
updater2p=1
```

My suggestion would be to append to that file when it's first created with

```
[Atmosphere]
fss0=atmosphere/fusee-secondary.bin
```

or similar, depending on what the Switch is running.

CTCaer, 2020-12-31:
What you suggest will need to search and parse files in order to find the correct one.
That's because hekate does not hardcode any paths that are not hekate's or crucial for functioning.
It's easy to just edit the file for that. Also it's a preferred process in order for user to learn about how the config works.

It's not user friendly, but that specific thing was never expected to be.
Reason why every guide and also the template in this repo can help you with.
Before 7.0.0, it would ask you to boot stock if no boot entry, but that's not possible anymore.

So, it's basically a process that user should learn and also create a config at least once.
That's the reason also why hekate never ships a config. It respects every single configuration and forces none.

closed 2020-12-31 by CTCaer

j-james, 2020-12-31:
All right, that's fair enough.