CTCaer/hekate #543: [BUG/VISUAL] All dump done window shows broken dump path

eXhumer, 2020-12-31:
https://github.com/CTCaer/hekate/blob/9d79f398972f856f1c10ebe4d943a89be9c1810a/nyx/nyx_gui/frontend/gui_info.c#L74

This line of code has `{emmc_sn}` hard-coded and doesn't attempt to replace it with the actual SN. As a result, the dump complete window always shows that hard-coded `{emmc_sn}`.

First noticed the bug while dumping fuel_reg.bin, but this affects any function that uses `_create_window_dump_done`.

[image]

eXhumer, 2020-12-31:
PR #543 created to fix this issue.

CTCaer, 2020-12-31:
That's not actually a bug.
From start it was intended that way in order to avoid touching the emmcsn_path_impl logic.
And normally when user asks what is emmc_sn or that strange number there, learns and then remembers that it's the eMMC serial number.

If for whatever reason the actual sn is wanted there I can refactor the whole function for that.
(The PR might get rejected because I already made some changes locally for the dumps that can create a conflict. Also the function is in need of a refactoring. It was written back when avoidance of string functions was wanted.)

eXhumer, 2020-12-31:
OK.

I'll re-implement the functions to work with upcoming changes in the PR. If you do not intend to use the provided solution, feel free to close both the issue and the PR.

CTCaer, 2021-01-03:
Completely refactored in https://github.com/CTCaer/hekate/commit/53b44a525d3a304c21c8de19bea0d939ae25f6f5
Issue will stay open until the release.

closed 2021-01-05 by CTCaer