CTCaer/hekate #548: Issues with SD or is it BUG

elozbay, 2021-01-03:
Hi guys!

Can someone please help me and guide me step by step, how to go about solving this. :(
Tried to install the new hekate and I get this error

[image]
[image]
[image]
[image]

impeeza, 2021-01-03:
Not a bug, please read the readme file. Almost sure your SD card is fault.  Update to the latest hekateand use a new SD fir testing

balika011, 2021-01-03:
Looks like an improperly seated or damaged sdcard reader.

elozbay, 2021-01-03:
> Not a bug, please read the readme file. Almost sure your SD card is fault. Update to the latest hekate and use a new SD fir testing

I put in another SD card and reformatted it and put in the new hekate (updated) and the stil same error :(

elozbay, 2021-01-03:
> Looks like an improperly seated or damaged sdcard reader.

I put in another SD card and reformatted it and put in the new hekate (updated) and the stil same error :(

impeeza, 2021-01-03:
Sounds like a fault card reader on the switch. search the net for solutions

impeeza, 2021-01-03:
Sure you will need to apply the solution on: https://gbatemp.net/threads/problem-with-sd-card-reader-seat-from-nintendo-switch-mother-board.552215/#post-8867689

Please sell help on the correct site

CTCaer, 2021-01-03:
That's a HW issue with your sd card reader's connector.

closed 2021-01-03 by CTCaer