CTCaer/hekate #550: Booting Atmosphere on stock firmware

myeuropeanPAL, 2021-01-05:
Hi, I am trying to setup a boot option for Atmosphere without emuMMC, but with emuMMC also present and enabled on the SD card as a file. I tried this config:

[Atmosphere (CFW)]
payload=bootloader/payloads/fusee-primary.bin
emummc_force_disable=1

[Atmosphere (emuMMC)]
payload=bootloader/payloads/fusee-primary.bin

The second option is working as expected, but the first boots also into emuMMC, if emuMMC is set to enabled.
The first option is working when emuMMC is disabled manually in Hekate.
This is with the latest Hekate, Atmosphere and original firmware. (January 2021)

Could you please advise how to fix this?
Thanks.

impeeza, 2021-01-05:
The issues section of the github is no the place to ask for help, please read the documentation, and if is no sufficient, go to the official support channels, ie, discord, wikis, forums, etc.

The issues section IS ONLY FOR reporting real issues on the program.

CTCaer, 2021-01-05:
You are effectively bypassing hekate booting with that.
What you want to do is not possible without using hekate to boot HOS/CFW.

closed 2021-01-05 by CTCaer

myeuropeanPAL, 2021-01-16:
The solution:

{---Nintendo Stock---}
[OFW (SYSNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
stock=1
icon=bootloader/res/ofw.bmp
{}

[Atmosphere (CFW)]
fss0=atmosphere/fusee-secondary.bin
emummc_force_disable=1
{}

{---Atmosphere Emu---}
[Atmosphere (emuMMC)]
payload=bootloader/payloads/fusee-primary.bin
{}