CTCaer/hekate #553: Failure to Launch Atmosphere

Arch9SK7, 2021-01-06:
This is in regards to the newest release of Atmosphere 0.17.0.

I CAN launch atmosphere from hekate bootloader menu but only if i choose the payloads option. 

If i choose the boot cfw sysnand or whatever it boots in OFW only.

This was tried using the SX chip on switch lite firmware 11.0.1
also tried using spacecraft-nx chip. Results same.

No error messages appear, it just fails to load the cfw.

urherenow, 2021-01-06:
Post the contents of hekate_ipl.ini.

And keep in mind, when using fusee, you are not using hekate at all.

Arch9SK7, 2021-01-06:
> Post the contents of hekate_ipl.ini.
> 
> And keep in mind, when using fusee, you are not using hekate at all.

[config]
autoboot=0
autoboot_list=0
bootwait=3
verification=1
backlight=100
autohosoff=0
autonogc=1
updater2p=1

{--- Custom Firmware ---}
[CFW (SYSNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
atmosphere=1
debugmode=1
kip1=noexs.kip1
{}

[CFW (EMUMMC)]
emummcforce=1
fss0=atmosphere/fusee-secondary.bin
atmosphere=1
{}

{--- Stock ---}
[Stock (SYSNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
stock=1
{}

The contents of my ipl :)

urherenow, 2021-01-06:
I've never actually seen somebody use emummcforce. What do you have in emummc.ini?

CTCaer, 2021-01-06:
You can't just boot OFW without stock=1 except if you hang the SoC and make it reboot.
Also why you have these there:
```
debugmode=1
kip1=noexs.kip1
```

Anyway, you can use `Logs ON` in Launch menu before you launch a boot entry and you will get a boot log.
Try to get that first. (You might need to take a video because normally it rolls very fast.)

Also 1emummcforce1 is fine and actually nice to have in emuMMC boot entry. It stops booting if emuMMC is not found or has issues, and thus protecting user from running sys CFW when it's not wanted.

The only thing that is not needed in these configs is `atmosphere=1` because the `fss0` implies that.
But doesn't change anything if you have it or not anyway.
(As for what I said above, I would remove these 2 lines from sys and check)

urherenow, 2021-01-06:
I didn't say it wasn't good/nice, I just said I haven't heard of someone using it. I asked about the .ini both because I was curious, and because the readme states it will throw an error if the .ini is missing. but it also isn't clear about a template and such for that particular .ini. Perhaps I need to read it over again.

Glitching-Queen, 2021-01-06:
> You can't just boot OFW without stock=1 except if you hang the SoC and make it reboot.
> Also why you have these there:
>
> ```
> debugmode=1
> kip1=noexs.kip1
> ```
>
> Anyway, you can use `Logs ON` in Launch [...]

I will do the logs and get back to you with that as for those other two lines I use those because it allows me to run Noexes debugger. Ill also remove the atmos line and see as well.

Glitching-Queen, 2021-01-06:
> You can't just boot OFW without stock=1 except if you hang the SoC and make it reboot.
> Also why you have these there:
>
> ```
> debugmode=1
> kip1=noexs.kip1
> ```
>
> Anyway, you can use `Logs ON` in Launch [...]

[image]
Sorry this is a tad blurred but this is what the log says. It indicates everything to be fine. I did remove the two debig and kip lines as well. 

Also @urherenow i am sure its OFW. The black atmosphere splashscreen shows up then soft reboots into OFW and album is automatically HOS album still.

Arch9SK7, 2021-01-06:
> You can't just boot OFW without stock=1 except if you hang the SoC and make it reboot.
> Also why you have these there:
>
> ```
> debugmode=1
> kip1=noexs.kip1
> ```
>
> Anyway, you can use `Logs ON` in Launch [...]

[image]
Sorry this is a tad blurred but this is what the log says. It indicates everything to be fine. I did remove the two debig and kip lines as well.

Also @urherenow i am sure its OFW. The black atmosphere splashscreen shows up then soft reboots into OFW and album is automatically HOS album still.

CTCaer, 2021-01-06:
If you see the b&w logo then it passed over hekate and you booted but something failed later.
It looks like you got a fatal but the fatal error payload was not able to show a screen or something and it rebooted without a full power off, which bypasses the glitch process and boots ofw. 
(Assuming you are really getting to OFW, because normally the first point of OFW/Stock vs CFW is that the b&w logo is the nintendo one.)

I would suggest to:
- Check the fatals folder for new files. You might have a new one that says what's wrong.
- Rename the atmo folder and drop a clean one for testing
- Run the archive bit fixer just in case that's the issue.
- Check that nogc is not the cause for that.

urherenow, 2021-01-06:
Ditto on the atmosphere folder rename. It could be very well be something you added to the contents folder. If you're setup for EdiZon SE, there is likely updated files that you need, since it requires a thing that was forked from Atmosphere, and also files forked from the vanilla noexes...

Arch9SK7, 2021-01-06:
OK so i followed all the suggestions. archive bit unset and set 0 for both.
I put a fresh clean atmosphere and still no resolve. it will show Atmos B&W then nintendo splash and boot OFW. tested with nogc on and off same results each time. as for fatals there was nothing in that folder pertaining to any errors.

CTCaer, 2021-01-06:
Did you also rename exosphere.ini?

CTCaer, 2021-01-06:
Wait, atmos b&w and nintendo switch?

Arch9SK7, 2021-01-06:
> Wait, atmos b&w and nintendo switch?

yup. it does both. and no i have yet to rename the exosphere

CTCaer, 2021-01-06:
Yeah what you boot is sys CFW. It's not ofw..
You can see it in the settings at the version info even.

closed 2021-01-06 by CTCaer

Arch9SK7, 2021-01-06:
> Yeah what you boot is sys CFW. It's not ofw..
> You can see it in the settings at the version info even.
I understand what you are trying to say but if I check the album there's no attachment to Homebrew if I check the system settings perversion info its just HOS Version info. Nothing with atmosphere. I wouldnt have made this an open issue if i didnt already know all this and check the info etc.

CTCaer, 2021-01-06:
You can't just see atmo b&w -> switch logo -> lockscreen/home and get OFW.

Arch9SK7, 2021-01-06:
> You can't just see atmo b&w -> switch logo -> lockscreen/home and get OFW.

https://youtu.be/O8klwc2arbI

Watch and you will see

CTCaer, 2021-01-06:
That's not the "nintendo switch" logo. It's the nintendo logo.
So atmosphere fatals after boot.kip is done and does a power off - reset combo. I can't help with that. Especially if doesn't save a fatal in sd card.

Arch9SK7, 2021-01-06:
Ok thank you.

CTCaer, 2021-01-06:
Btw because I saw you are using a pack, the new atmosphere folder was supposed to be pasted from the atmosphere release, in case you didn't try that.

Arch9SK7, 2021-01-06:
> Btw because I saw you are using a pack, the new atmosphere folder was supposed to be pasted from the atmosphere release, in case you didn't try that.

yeah i did try and was no different sadly.