CTCaer/hekate #556: [Feature Request] Add support for using joy-cons wirelessly

amblerdesign, 2021-01-07:
I absolutely love the USB gamepad emulation. It solved all my input lag and connectivity issues when playing PC games with the controllers. The only thing I regret is that I have to use the joy-cons when connected in the rails, and of course its a bit cumbersome holding the entire console.

I hope you may add this in future :)

Masamune3210, 2021-01-07:
impossible in the current environment at least, bootloader is minimal hardware and software so no access to bluetooth or other wireless systems

CTCaer, 2021-01-08:
Well it's not impossible but there's no way I will port BT drivers and BT stack only for that.
(And wifi is even worse on that matter. Needs PCIE and wifi drivers and a supplicant stack.)

closed 2021-01-08 by CTCaer