CTCaer/hekate #558: Genuine Boot Option

H4X0R46, 2021-01-11:
Is there any chance you could consider adding a genuine boot option? Mariko users right now don't have any option to boot the system cleanly for online play and sysnand updates when using Atmosphere, since Atmosphere is so new on Mariko.

urherenow, 2021-01-11:
Can you not already do that yourself in the .ini? I don't have a mariko unit, so I'm not sure.

emummc_force_disable=1
stock=1

@CTCaer just noticed that the above disable flag is in the readme list, but the description for stock says emummc_force_disabled=1.

H4X0R46, 2021-01-11:
> Can you not already do that yourself in the .ini? I don't have a mariko unit, so I'm not sure.
> 
> There are stock setups in the template, even for using a stock emummc. You likely want [Stock ALL FW]
> 
> @CTCaer just noticed that the emummc_force_disable=1 flag is in the readme list (and template), but the description for stock says emummc_force_disabled=1.

Yeah you can, that works, but my concern is if it's safe enough to use for legit online play and sysnand updates without a very high ban risk. Is it "clean" enough?

masagrator, 2021-01-11:
Genuine boot rrquires locking glitching until reboot.
Using stock in hekate won't boot 1:1 as OFW.
This issue seems more related to spacecraft than Hekate as spacecraft needs function for this to be implemented and then Hekate could take advantage of it.

H4X0R46, 2021-01-11:
> Genuine boot rrquires locking glitching until reboot.
> Using stock in hekate won't boot 1:1 as OFW.
> This issue seems more related to spacecraft than Hekate as spacecraft needs function for this to be implemented and then Hekate could take advantage of it.

I've been hesitant on flashing spacecraft honestly. I'm still on SX firmware for my Core. But I'm thinking if TX was able to do it, shouldn't it be easy enough to implement into Hekate? Or did TX do underhanded proprietary crap like they're known for?

BlackRabbit22, 2021-01-12:
@masagrator 
 
 According to balika011, it is an issue that is related to hekate and not spacecraft-nx here in this [comment](https://github.com/Spacecraft-NX/firmware/issues/8#issuecomment-755237689).

H4X0R46, 2021-01-14:
> @masagrator
> 
> According to balika011, it is an issue that is related to hekate and not spacecraft-nx here in this [comment](https://github.com/Spacecraft-NX/firmware/issues/8#issuecomment-755237689).

That's good to know! Since atmosphere is so new on Mariko, and we don't actually have an easy way to boot 100% stock since our exploit is a soldered modchip, a genuine boot option would be ideal! 👍

CTCaer, 2021-01-14:
It's not an issue with hekate.
hekate will never get distributed with copyrighted data such the one needed for the "genuine boot".
Now if you really want that, it's purely a modchip/fw issue. And ask them to fix that.

As for hekate, best you can get is the OFW option reinstated for patched consoles.
And that's only if you have proper secondary BCTs.

Stock option from hekate is always the better option if you don't want to burn fuses.
That option only replaces ninty's secmon with exosphere and nothing else for Marikos. Everything else is kept official.

closed 2021-01-14 by CTCaer