CTCaer/hekate #559: Pkg1 decryption failed! Is BEK missing

shoyrock, 2021-01-11:
hey I've added hekate_keys.ini to root/bootloader/hekate_keys.ini

[Prod]
mariko_bek = 
mariko_kek = 
package2_key_0a =
master_key_0a =

but I'm still getting BEK missing, am I doing something wrong, or am I missing something

urherenow, 2021-01-12:
Don't have a mariko but a quick Google says:

keys.
# Together the Mariko BEK and KEK are enough to derive all current content keys and all
# content keys in the forseeable future except for Erista package1.

So... do you have the needed package1 key? Most likely different from the easily obtainable erista package1 keys...

shoyrock, 2021-01-12:
> Edit: Misread what I found on Google. I don't have a Mariko unit so I can't help.
> 
> It may help to edit your post to include your versions and methods (ie. spacecraft-nx version...? And Hekate version...? Switch firmware version, Atmosphere version too. Except for Switch fw, every one of those should be the latest, and not from some package like sdfiles.zip...

I've added more into so I hope that helps

urherenow, 2021-01-13:
Just to be sure, your Hecate_keys.ini has the ACTUAL keys in it, and not just the = xxxxxxxxxxxxxxxx followed by the sha256?

From what I've read, it really is better for you to flash spacecraft, even if you figure out your current problem. Also, GitHub is not a support forum, so you're better off asking for help on gbatemp, or even discord or Reddit, for that matter.

shoyrock, 2021-01-13:
[image]
I don't have the USB flex needed to flash the chip so there's no way for me to use spacecraft. just so you know I'm not the only one that has this problem because they got someone else to do the install for them. it wasn't needed b4 now so it wasn't a problem b4 now

paulzheng136, 2021-01-13:
is it happened only in mariko's

paulzheng136, 2021-01-13:
> I've added more into so I hope that helps

why don't you replace boot.dat as sx gear 1.1's.   it needn't to flash sx core's firmware in spacecraftnx

shoyrock, 2021-01-14:
> why don't you replace boot.dat as sx gear 1.1's. it needn't to flash sx core's firmware in spacecraftnx

that's what I'm using now

CTCaer, 2021-01-14:
hekate_keys.ini is not something that hekate actually supports. I don't know what you are talking about.
Also refrain from illegally posting even a single part of the keys.

closed 2021-01-14 by CTCaer