CTCaer/hekate #560: Issue with Migrate emuMMC funtion in Mariko

Replibac, 2021-01-12:
Hekate 5.5.2
Nintendo switch Mariko on 10.2.0
SX Gear to launch

I would like to share this with you:

Without previous emunand created, only hekate and sx gear to launch it in the SD I have found this bug:
- Entering via tools I have create a partition of 29gb in the SD
- Then, entering via emuMMC I have push in migrate emuMMC > Fix RAW
- Exit and enter again to emuMMC, and hekate shows that I have activated emunand.
- Instaled atmosphere and in the firmware version shows that I have emunand installed ( without emunand files )

link to video ( coments in spanish, sorry ):
https://drive.google.com/file/d/116w4Ju5jBbIbEldvLd1J3lfgvn3ajHWD/view?usp=sharing

paulzheng136, 2021-01-13:
you needn't to create a partiton before migrate emummc,   use this  functions directly, there are two different formal in hidden partition emunand between SXOS  and Hekate.   if you made hidden emunand in SX before.   just use migrate emummc enough.

Replibac, 2021-01-13:
Thanks for answer.

I wil try to explain how I have found that. The first test was this:
I have a main sd with sx os emunand in hidden partition in my SD and I want to leave this one intact and do some tests in another SD, so:
- I have extracted the emunand using emutool. There is not a possibility to create from hidden sx os partition to atmosphere hidden partition directly, so I have transfer from SX OS hidden partition to atmosphere visible files.
- Then, I have formated and created new partition (29gb) as you see ( I had some problems with the partitions letters, but solved )
- After that, I have transfered the converted atmosphere file into this partition (29gb) and push "Migrate emuMMC", but only it recognized "RAW type", so I have pushed there.
- When I check the emummc.ini file, see taht the path was to primary partition RAW folder and not to the 29gb partition folder.
- I deleted the files in partition of 29gb and I see inside hekate that shows that the emunand is enabled, and cheking in atmosphere FW see the E letter at the end, without emunand files.

So, I am not sure if it is an issue or simplement a casuality. So I have opened this issue, because doing that we think that we are runing in emunand and this is not true, we are runing in CFW on sysnand, dirtying it.

paulzheng136, 2021-01-13:
well,  i am not a techguy.  just discuss it as amateurs.
you have extracted the emunand as visible files right?  around 29gb on your desktop.    then you can rename them as hekate's type copy on your SD card according to root: emummc/SD00/emmc/ ,  don't forget to create a empty file named "file_based" in emummc/SD00/.  
payload as hekate.bin  then you know what to do next step?

Replibac, 2021-01-13:
you have extracted the emunand as visible files right?
- yes
then you can rename them as hekate's type copy on your SD card according to root: emummc/SD00/emmc/ , don't forget to create a empty file named "file_based" in emummc/SD00/.
- no, I have use again emutool to convert from atmosphere file to atmosphere hidden partition and then select the partition of 29gb in my SD, trying to leave it in hidden mode, but nothing, so I can see the name of the files folder

paulzheng136, 2021-01-13:
i compared the 2 hidden mode with  hekate and sx  via diskgenius 1 years ago.    the sx's  is primary partiton on SD card ,  but ams's is in rear partition.  i don't know why cfw can mount it that.  so i decidedd ingore this kind of  hidden emunand mode.    
you can define emunand's path in hekate_ipl.ini if use fss0 to launch emummc.

CTCaer, 2021-01-14:
> Hekate 5.5.2
> Nintendo switch Mariko on 10.2.0
> SX Gear to launch
>
> I would like to share this with you:
>
> Without previous emunand created, only hekate and sx gear to launch it in the SD I have found [...]

That's not an actual issue. hekate checks for the emuMMC GPT explicitly.
Having that button enabled is because you had emummc at the same offset in the past.

I also won't add more validation checks because they would be changes to "fix" an intentional wrong behavior from user. (Repartitioning sd card with emuMMC partition and not creating an emuMMC).
And not only that but they could still succeed without a wipe.

closed 2021-01-14 by CTCaer