CTCaer/hekate #563: [bug] Mariko Users Black screen on launch

squee666, 2021-01-14:
As of 5.5.3 Hekate does not work for mariko units.

Launching the payload in any form results in black screen and requires hard reboot.

Tried launching

- through sxos payloads menu
- through sxgear directly so no keyslots wiping
- through spacecraft
- without bootloader folder

3 different mariko units all the same result.

previous version 5.5.2 still works fines on all

thanks again for time

CTCaer, 2021-01-14:
```
through sxos payloads menu
through spacecraft
```
Doubt about these.

As for that gear thing it corrupts the payload if some specific checkboxes are crossed.
You probably tried the other options after you used the gear one which corrupted the payload and hekate created a corrupted update.bin, which then any subsequent try was chainloading that.

Anyway, release was updated to mitigate that.
Redownload, delete `bootloader/update.bin` and use the updated `hekate_ctcaer_5.5.3.bin`

closed 2021-01-14 by CTCaer

squee666, 2021-01-14:
I actually tried sxos/payloads first

before trying the rest and unless you updated it 40 minutes ago, its the same result