CTCaer/hekate #566: Update Hekate without updating Dongle

victorgneves, 2021-01-17:
It wouldn't be possible for me to update my Hekate in SD, load the Dongle with an outdated version of hekate and he check the version in SD, load it and already show me an option to copy it and update my Dongle, without having to update it manually?
Would Dongle allow this recording made by Hekate?

Another idea would be a mini bootloader that never changes on the Dongle, and that it would quickly load the full version of the hekate found in the SD.

Maybe there is already a function like that and I don't know.
If it already exists, I'm sorry. kkk

Thank you for the excellent work you have done.
Sorry for my English, Google translator.

urherenow, 2021-01-17:
The check->load from SD is certainly possible. That's how the SX loader works, more or less. I'm all for this idea.

Updating itself... not sure it's possible on most (all?) dongles. The main dongle in use uses a microSD cable for updating the payload. The C port does the injection. Not sure I want to see what happens if I plug that end into my PC.

Masamune3210, 2021-01-17:
The R4 dongle updates through a special mode over micro usb, so it would also be a case of supporting all the different dongle types that are popular

impeeza, 2021-01-18:
For the R4 and clones dongles, is as simple as connecting the dongle to the
computer using micro USB Cable and the computer sees it as a normal USB
Drive and you can replace the file on the corresponding folder.

Masamune3210, 2021-01-18:
I know for a fact that you have to put the R4 into a special mode first, but either way that isn't the point. The point I was making is that not all dongles expose anything to the switch, other than the initial payload push

victorgneves, 2021-01-18:
> I know for a fact that you have to put the R4 into a special mode first, but either way that isn't the point. The point I was making is that not all dongles expose anything to the switch, other than the initial payload push

Exact. I'm not sure, but I believe most dongles would not allow a replacement of the payload with USB-C, but I would have to test it.

Anyway, a micro payload.bin that when loaded, loads "SD / payload.bin".
Even if there is a 0.5s delay.

Maybe there is already someone who did one like this and I don't even know.

SXOS already does that, but it looks like it doesn't start another payload, but boot.dat.

impeeza, 2021-01-18:
The R4 Dongles need to be updated with a special UF2 file containing the payload, pressing two times the button when connected to the computer using the MICRO USB port.   Or you can put a firmware which search for payload.bin on  the Switch's SD card at argon/payload folder. (http://www.r4i-sdhc.com/R4SDongles.asp).

I use a clone called RCM Loader and is very simple, you can update the payload coping it to dongle directly as an USB drive

impeeza, 2021-01-18:
This is no a Hekate issue so please close the issue and if need furter help use the propper suppor channels

victorgneves, 2021-01-18:
> The R4 Dongles need to be updated with a special UF2 file containing the payload, pressing two times the button when connected to the computer using the MICRO USB port. Or you can put a firmware [...]

yes, i also have RCMLoader, and always update by micro-USB.
I just have to do this every time the hekate updates, and always taking care not to forget.
In 1 year it's about 30 kk.
It would just be a question.
But it is not the end of the world. kkkk
Let's look forward to the new updates.
It is always fun to read the release notes.

CTCaer, 2021-01-19:
The whole convo seems that it went to random thoughts and assumptions.

hekate always used **/bootloader/update.bin** for that reason (since 2.5 years now).

closed 2021-01-19 by CTCaer