CTCaer/hekate #568: Switch crushing after moving game to NAND from settings in file-based EmuNAND

rashevskyv, 2021-01-23:
## Bug Report

### What's the issue you encountered?

Installing game from eShop, then goes to settings and moving game to NAND with internal HOS method. 
File-based EmuNAND, on **Samsung EVO 128Gb microSD**, formatted with **hekate 5.5.3**. Booted from `hekate =- payloads =- fuse-primary.bin`
Using `atmosphere-0.17.1-master-49f07b06+hbl-2.4.0+hbmenu-3.4.0` from github. I have unpacked `hekate_ctcaer_5.5.3_Nyx_0.9.9.zip` to the root of sd, then unpacked `atmosphere-0.17.1-master-49f07b06+hbl-2.4.0+hbmenu-3.4.0.zip`, renamed `hekate_ctcaer_5.5.3.bin` to `payload.bin` and put `fuse-primary.bin` too `bootloader =- payloads`

My modchip works on Spacecraft-NX. 

### How can the issue be reproduced?

1. Format SВ with hekate 
2. Remove all files from SD
3. Unpack `hekate_ctcaer_5.5.3_Nyx_0.9.9.zip` to the root of sd
4. Unpack `atmosphere-0.17.1-master-49f07b06+hbl-2.4.0+hbmenu-3.4.0.zip`  to the root of sd
5. Renam `hekate_ctcaer_5.5.3.bin` to `payload.bin` 
6. Put `fuse-primary.bin` too `bootloader =- payloads`
7. Launch console
8. `Payloads =- fuse-primary.bin`
9. Install game from eshop
10. Go to **Settings =- Data Management =- Move Data Between Console/microSD Card =- Move to System Memory =- Choose game =- Move Data =- Move**
11.  After approximately 1 minute system will crash with error provided on screenshot below 

[image]

### Crash Report

[atmosphere.zip](https://github.com/CTCaer/hekate/files/5860421/atmosphere.zip)

### System Firmware Version

11.0.1

### Environment?

Using `atmosphere-0.17.1-master-49f07b06+hbl-2.4.0+hbmenu-3.4.0` from github. I have unpacked `hekate_ctcaer_5.5.3_Nyx_0.9.9.zip` to the root of sd, then unpacked `atmosphere-0.17.1-master-49f07b06+hbl-2.4.0+hbmenu-3.4.0.zip`, renamed `hekate_ctcaer_5.5.3.bin` to `payload.bin` and put `fuse-primary.bin` too `bootloader =- payloads` without additional homebrew, patches or other 

Tested on Switch Light and big Switch on Mariko

rashevskyv, 2021-01-23:
With fss0 (`fss0=atmosphere/fusee-secondary.bin`) I got this error: 
[image]

[fatal_reports.zip](https://github.com/CTCaer/hekate/files/5860451/fatal_reports.zip)

urherenow, 2021-01-23:
Pretty good report but 1 more thing is needed please. Look at your SD card info in nyx, and post a pic of that.

rashevskyv, 2021-01-23:
[image]
[image]
@urherenow is this exact information that you asked for?

urherenow, 2021-01-23:
Yesterday it is, thanks. It's basically to ensure you didn't end up with a fake card, and that your SD slot isn't broken and stuck in a 1bit mode. It appears to be a good SD to me.

CTCaer, 2021-01-24:
You are reporting that in the wrong repo.

Also that's probably the timeout issue on file based when both emmc (SD) and SD are 100% utilized.
Imo, I find 0 reasons to move a game **from SD to SD**. If you want to save space, just make the emummc USER partition smaller.

closed 2021-01-24 by CTCaer

rashevskyv, 2021-01-24:
Okey, I duplicate report to emummc repo, sorry

rashevskyv, 2021-01-25:
> I find 0 reasons to move a game from SD to SD
I have this problem even when I installing game on NAND from eshop. So it is not about moving.