CTCaer/hekate #569: NXBOOT failed to mount Emmc

Blasphemy1823, 2021-01-28:
Machine: Erista Patched Chipped
Hekate version: 5.5.2
Atmosphere: 17.1
Boot sequence: launching and delivering payload via Hekate. 
emummc: tried with and without

question 1: Does the screen of the Emmc status show that the Emmc chip might be dying because it is unable to detect the remaining life. What is A and B?

question 2: i am getting failed to mount Emmc when launching atmosphere. Could a faulty or dying Emmc chip cause this?

many thanks for your help

[image]
 
[image]

CTCaer, 2021-01-28:
You are reporting this in the wrong repo.
Your eMMC clearly works and is fine in hekate.
It doesn't work in the payload you are launching. If this can't be fixed there, then you can try booting via hekate.

closed 2021-01-28 by CTCaer

Blasphemy1823, 2021-01-28:
Hi i am booting via Hekate.. below is my ipl.ini

FSS0 brings me to to a black page and Primary brings me to the screen that says failed to mount emmc..

[config]
autoboot=0
autoboot_list=0
bootwait=4
verification=1
backlight=100
autohosoff=0
autonogc=1
updater2p=1

{--- Stock ---}
[Stock (SYSNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
stock=1
{}

{------ Atmosphere ------}
[Atmosphere EMU]
payload=bootloader/payloads/fusee-primary.bin
emummcforce=1
{}