CTCaer/hekate #572: Autoboot fails

dza89, 2021-01-31:
hekate: v5.5.3
firmware: 11.0.1

Whenever i enable autoboot, I can boot from Hekate to CFW sysMMC.
When I then restart it boots to stock sysMMC.
When I enter RCM mode and send a payload, it boots to sysMMC.

If I remove the SD card and change the autoboot value to 0, I can enter Hekate and CFW again.
Tested on two switches, with two different sdcards, both the same result.

Bootloader tree:
```
├── hekate_ipl.ini
├── ini
├── nyx.ini
├── patches.ini
├── payloads
│   └── fusee-primary.bin
├── res
│   ├── icon_payload.bmp
│   └── icon_switch.bmp
├── sys
│   ├── emummc.kipm
│   ├── libsys_lp0.bso
│   ├── libsys_minerva.bso
│   ├── nyx.bin
│   └── res.pak
└── update.bin
```

content hekate_ipl.ini
```
[config]
autoboot=0
autoboot_list=0
bootwait=2
backlight=100
autohosoff=0
autonogc=1
updater2p=1
bootprotect=0

[CFW - sysMMC]
fss0=atmosphere/fusee-secondary.bin
kip1patch=nosigchk
emummc_force_disable=1

[CFW - emuMMC]
fss0=atmosphere/fusee-secondary.bin
kip1patch=nosigchk
emummcforce=1

[Stock - sysMMC]
fss0=atmosphere/fusee-secondary.bin
emummc_force_disable=1
stock=1
```

How can I extract some logging? The only toggle I see if in Launch and that flips back after I leave the Launch Screen.

ELY3M, 2021-02-02:
the autoboot can only boot one thing.   take a look at autoboot_list=0    you are booting cfw on sysnand.    you need to change to autoboot_list=1  if you want to boot emummc.

dza89, 2021-02-02:
I want to boot CFW sysMMC, but I'm booting to Stock. 
When I change autoboot_list to 1, I also boot to stock.

dza89, 2021-02-07:
Fixed my own mistake

closed 2021-02-07 by dza89