CTCaer/hekate #573: [ENHANCEMENT] NYX scripting

urherenow, 2021-02-08:
Would it be possible to recycle GodMode9 scripting code for use under nyx? Even a stripped down version for file commands? moving, copying, renaming, etc.

CTCaer, 2021-02-08:
Dunno what this scripting is but what I understand is that you want sth like #254 + BIS

urherenow, 2021-02-08:
No,I would like to be able to run scripts (like a batchfile) as the very first use I have in mind involves moving and renaming 7 files at the same time. Yes, a file manager could work, but it would still not be as convenient.

TBH, I've trying to carefully avoid stating the exact immediate purpose because it's something I'm fairly certain you'd flat out say no to. But running scripts does have other uses, as shown on the 3ds.

CTCaer, 2021-02-08:
Then no. Nyx's scope is not to become a shell.
It's supposed to do specific things and try to do them well.
Extra 3rd party stuff can be added via modules I guess. When I actually finalize the design of that at least..

Anyway, you can do all that fairly easier with other script tools or via pc.

closed 2021-02-08 by CTCaer

urherenow, 2021-02-08:
Via PC destroys the portable aspect, and the files that need moving cannot be touched within HOS. It's bare metal or bust.