CTCaer/hekate #574: Failure to backup BOOT0 & BOOT1

SpectralSlog, 2021-02-08:
When I try and backup BOOT0 and BOOT1 I get this error
[image]
I also get a similar error when trying RAW GPP
[image]

Please help! Thanks.

CTCaer, 2021-02-08:
Go to partition manager and the moment the pop up that has your partition table shows up, take a screenshot (3 finger tap or screenshot button on jc) or a photo.
Up it here

SpectralSlog, 2021-02-08:
@CTCaer This?
[image]

CTCaer, 2021-02-08:
You didn't create your emummc via hekate and thus it has wrong MBR entry.
emuMMC partition type is not exFAT (x07), it's xE0.
Correct that with a hex editor (it's in sector 0) and it will be found.

closed 2021-02-08 by CTCaer