CTCaer/hekate #576: Error : Failed to run Sept, Main BCT is improper"

kaydayyy, 2021-02-10:
Hello, yesterday I updated Hekate and after booting up the sysnand or emunand I get that error, and I cant find any fix, what I have to do? please help :(

"Now an error will show up that instructs user to run sept with proper BCT at least once, in order to cache keys in EKS"

How do I fix that? sorry for being that ignorant :(

urherenow, 2021-02-10:
How exactly did you "update hekate"? Hekate isn't a CFW, it's a bootloader. It sounds like you didn't grab the latest Atmosphere, drag the ENTIRE contents of the .zip onto your SD card (do you even have a sept folder?) and overwrite everything it asks you.

If you did, inject fusee-primary.bin once to load vanilla Atmosphere. Then try hekate after that.

CTCaer, 2021-02-10:
The modchip you run changed the main BCT in BOOT0.
You will need to restore that at least once before launching sept.

closed 2021-02-10 by CTCaer