CTCaer/hekate #579: sd and emmc data <@LBA 00008000>, do not match

iSkrrtt, 2021-02-17:
Says “verification failed, please try again... Failed!”

I installed correctly and it gets stuck at 6%/3m 41s. 

boot0 and boot1 backed up fine. 

Any help would be appreciated
Thanks!

[image]

CTCaer, 2021-02-19:
Your sd card is probably fake.
If you want to validate that easily, send me a screenshot or photo from Console info -> SD card.

CTCaer, 2021-02-28:
No reply from user so I assume they got a proper sd card

closed 2021-02-28 by CTCaer

Masamune3210, 2022-03-21:
Might try it with a fat32 FS, exfat is known to have gigantic issues on switch when it comes to HOS so maybe card FS damage?

ace90249, 2022-10-07:
[image]
I
[image]
 Have 
[image]
same issue. Please help...

ace90249, 2022-10-07:
[image]
And mybother Switch has this issue if u know anything about this mess lol. Thank you

impeeza, 2022-10-07:
You just open several reports, and all of then indicate you have a Corrupted SD card.