CTCaer/hekate #580: Touch Panel Error

Wat6656, 2021-02-18:
getting this error:
Model: Unknown Contact me!
ID AB5AC7EE (Unknown -Error)
FTB ver: E59B
FW rev: 75EB

CTCaer, 2021-02-19:
Interesting.
You can reliably get the same codes? Like if you re-enter the hw & fuses menu, you are still seeing AB5AC7EE, E59B, etc?

Wat6656, 2021-02-19:
You right the codes change. Now ID 0E30EF59
FTB ver: 1BED
FW rev: 54FA

Codes only change when I close Hekate and reboot. Should have clarified the touch screen does not work and unsure of its overall status, also the console can’t boot ofw or cfw and I am trying to unbrick it.

CTCaer, 2021-02-19:
So touch IC actually returns info but it's bad info.
And that info is probably stack garbage in the touch IC that gets random values everytime it gets powered on.
(Reason why it only changes if you close hekate and rerun).

What I assume happens is that it's missing the customized fw. The only way to flash it currently is by running HOS.
So until you are able to enter OFW or CFW you won't be able to use touch or get proper info in hekate.

closed 2021-02-19 by CTCaer