CTCaer/hekate #581: Option in settings to hide SysCFW boot from launch menu

RomEz10, 2021-02-18:
I feel like having the option to hide it will help prevent some unfortunate clicks and most users probably either care about booting to syscfw or really don't and it isn't prune to change frequently. If you don't want to implement that maybe just reordering the launch menu because the syscfw boot is overlapping with main menu launch icon. I know these might seem miniscule but I am really worried about this every boot, also, being miniscule probably means it's easy to implement.
Thanks!

impeeza, 2021-02-18:
I am wrong or this is configurated by user changing hekate_ipl.ini?. So you can remove the SysCFW entry on that file

RomEz10, 2021-02-19:
> I am wrong or this is configurated by user changing hekate_ipl.ini?. So you can remove the SysCFW entry on that file

thank you for your answer, sorry for bothering you for nothing.

closed 2021-02-19 by RomEz10