CTCaer/hekate #584: Emunand unusably slow when resized

WestwardFilly5, 2021-03-03:
Not sure if this is the place to mention this but HOS becomes painfully slow when using a resized partition created through hekate, and worsens over time.(minutes to get past the Nintendo logo launching games and freezing when streaming assets after a week)

I've tried 10,12 and 16 GB sizes all began to show this behavior after a few days of use.
I've also tried manually reformatting the "sd card" partition to use 32k cluster size as the default through the partition manager is 64k(?) with no improvement.

This only effects resized partition based emunand, All my testing was on Erista 11.0.1/AMS18.0/Hekate 5.5.4

Older thread detailing similar experience
[https://github.com/eliboa/NxNandManager/issues/15](url)

CTCaer, 2021-03-03:
What sd card you have?

WestwardFilly5, 2021-03-03:
A genuine Sandisk Ultra 400GB Micro SDXC UHS-I. I also have a 64gb Sandisk ultra I could try

CTCaer, 2021-03-03:
Do this simple test:
1. Use hekate's part manager and set emuMMC to 4GB.
2. Mount that card to PC with a good usb3 sd card reader
3. Format **both** partitions (hos and emu) as **exFAT 128KB**
4. Get crystaldiskmark and set [1] x [512GiB]

5. Select the first partition and run **All**
6. Screenshot results and do the same for the tiny partition
7. Post the 2 screenshots here

WestwardFilly5, 2021-03-04:
I couldn't workout how to directly format the small partition through diskpart so I used Disk Management to delete and recreate the volume, hopefully that's okay

Cardreader is the internal reader on Surface Tablet
[image]

CTCaer, 2021-03-04:
Can you re-run the test with 2x1GiB for both?
The write Q1T1 doesn't make sense.

WestwardFilly5, 2021-03-05:
[image]

CTCaer, 2021-03-05:
My hypothesis was that your sd had a really slow nand paired to that offset.
But the above screenshot invalidates that.

I can still bet that if you put the emummc partition before that offset in that sd will solve the issues.
And I believe that the controller changed the interleaving in order to give perf there after so many writes.

So in order to find what's happening with your Sandisk Ultra 400GB, try to create a resized emummc without partitioning/formatting the sd card again. So as it is, just go to emuMMC and choose create.
Test the speed and report back.

In case it's still slow and you are willing to test a custom nyx that rearranges the order, dm me in retronx discord.

mistervampi, 2021-03-07:
I have the same issue with 2 SD. But only happens in <29 GB emuMMC partitions created with Hekate.
Don't know if this info will help but I've created a 4GB emuMMC image on NXNandManager and works fine.

CTCaer, 2021-03-07:
File based (image) is written as a file to where your fat partition has free space, so it makes sense that it doesn't have that issue.
What we are trying to check here is why that card behaves like that but the bench does not show the issue.

For example I have 3 samsung evo plus and 1 sandisk extreme plus. None of them has that issue with a emuMMC part of 4GB.

But anyway, I made an offer above.
```
So in order to find what's happening with your Sandisk Ultra 400GB,
try to create a resized emummc without partitioning/formatting the sd card again.
So as it is, just go to emuMMC and choose create.
Test the speed and report back.

In case it's still slow and you are willing to test a custom nyx that rearranges the order, dm me in retronx discord.
```
Anyone that can reliably produce the issue and is willing to play with that can dm me.

DakuTree, 2021-03-12:
Have noticed this exact same issue after moving to an 8GB Partition EmuMMC. Switch menu feels slow, loading feels slow, and game music has a bad habit of stuttering/cutting out.
Could replicate the audio cut-out/delay issue 100% of the time when playing through the first stage of Mario 3D World, as well as the intro cutscene of Xenoblade Chronicles 2 if that is any help.

Tried to reformat card, repartitioned to 12GB, and tried other MicroSDs with no luck (1x 128GB UHS-I/U1 & 1x 128GB UHS-I/U3).

Moved to the default 29GB size and all problems appear to have gone.

Ammako, 2021-03-14:
I'm pretty sure this is either an Atmosphere issue, or an [emuMMC](https://github.com/m4xw/emuMMC) issue

https://github.com/eliboa/NxNandManager/issues/15

darkcloud545, 2021-04-17:
Same thing happens with my Lexar 633x UHS-I, Sandisk Ultra 128 GB UHS-I, Samsung EVO Plus U3 UHS-I MicroSDs.
Whenever I create a resized emuMMC with less than 29 GB, the switch menu as well as the tesla overlay loads slow. Could replicate with MH:Rise menu music cutting off. I tried running in sysNAND and the issue was gone. The issue also resolves with the standard size of 29 GB for the emuMMC partition.

Mewthree3, 2021-06-01:
Same thing happens on my Sandisk Ultra 200GB. I made a 5 GB emunand, and things run slow.

Yoti, 2021-06-30:
Same for me. SanDisk Switch 64 GB and Samsung EVO Plus 64 GB. Unpatched Erista, 11.0.1, latest AMS and Hekate.

mid-kid, 2021-07-22:
Same here, tried creating an 8GB emunand on a Sandisk Ultra 128GB. Had a similar issue on my last SD card.

FelicitusNeko, 2021-08-14:
Same deal. Samsung EVO 256, brand-new 4 GB emuMMC.

CTCaer, 2021-08-28:
Issue found.

On used up eMMC where the resized emuMMC would become smaller than the used size of eMMC, there was a chance to have Nand patrol point to out of bound addresses.
That was causing the SD card to time out and take seconds to get back.

That is now fixed and latest Nyx, when creating a resized emuMMC now clears Nand Patrol and effectively it has the same speed with a full sized one.
So anyone that wasn't to try it, just create/re-create the resized emuMMC in latest Hekate/Nyx.

closed 2021-08-28 by CTCaer

mid-kid, 2021-08-28:
Thanks a lot for figuring this out!
Is there any way to resize an existing emunand without making a new one from sysnand with Nyx? If not, can I clear Nand patrol with NxNandManager, or is it some file I can manually go and delete from the USER partition?

CTCaer, 2021-08-28:
If the existing is a full sized emuMMC on a partition, it's not so simple.
Also if it's full sized you don't need to clear this.

If you already have a resized partition you can recreate it (no need to repartition and reformat). Or just boot and nand patrol will be cleared automatically if you use hekate to boot.
And because I doubt you actually used it if it was the case, it's simpler to recreate it. It takes a minute or two to do so.

mid-kid, 2021-08-28:
I meant that I already have a full-size emummc. I was just wondering if it could be resized while fixing the nand patrol thing you mentioned, but if it's fixed when booting with hekate, I guess that's enough.
Thanks!

Mewthree3, 2021-08-29:
Thanks a lot for figuring this out! I don’t use an emuMMC (already got banned because ISP blocked 90DNS use), but this is a really great help. Thanks for all the great work!!⚡️⚡️

t33st33r, 2022-06-19:
I found this solution by accident and I am happy to find it but... I have a question about this:
I made a backup of my sysnand previous to this fix. This image has the above mentioned problems (stuttering on games and sometimes in HorizonOS, audio issues, etc). I used Haku33 to reset logs on my sysnand, so one solution is making the backup again from sysnand, so my question is: Is threre a way to fix the backup image without needing to back it up again from original?

Thanks in advance.

PS: English is not my native language. Please, excuse my low level english.

mid-kid, 2022-06-19:
I would love to know this as well, as the last time I looked into this, NXNandManager had not yet implemented the fix for resizing emunand, and I've still not seen anyone confirm hekate does it automatically for you.