CTCaer/hekate #590: Touch screen doesnt work (switch lite)

lurxyboy, 2021-03-28:
Hi,

I have a swich lite with SX lite modchip installed, Ive reflashed the chip and Im now using hekade and atmosphere instead of SXOS.

the problem is that my touch screen stoped working and I cant access the calibration option in Nyx/Hekate.
I know that the joycons work on the regular switch to navigate in hekate without the touch screen but not on the switch lite.

is there a way to start the calibration process without beeing able to navigate to it or can you make the buttons work on switch lite?

I've made a workaround to boot directly to CFW with fusee_primary.bin as payload so I can access the system - I just dont have a working touch panel.

hope that you can help me out mate!

impeeza, 2021-03-28:
Your problem dound like a hardware failure ;(

CTCaer, 2021-03-28:
Press and hold `VOL UP + VOL DOWN` and inject hekate. And keep pressing them until you see hekate home menu.

closed 2021-03-28 by CTCaer

lurxyboy, 2021-03-28:
> Press and hold `VOL UP + VOL DOWN` and inject hekate. And keep pressing them until you see hekate home menu.

That doesnt do it, when I inject with hekate bin I cant get past this menu because the touch doesnt work.

Is there another way to start the 3rd party calibration of the digitizer?

[image]

CTCaer, 2021-03-28:
That does the calibration. If it still doesn't work then your touch panel is broken or the connector is not seated well.
If you still want to use hekate to boot CFW, you can edit the ini manually and enable autoboot.
As it seems, no one actually cares to decode the protocol like I did for normal Joycon. And I have no mariko console anyway.

lurxyboy, 2021-03-28:
Am I supposed to see anything while booting up pressing volume + and -?

I dont really get from what screen I should start pressing and holding down the buttons.

I've done it like this:
1. Switch is turned off
2. Press and hold vol + and -
3. Boot up the switch and inject hekate, still holding vol buttons
4. Hekate boot screen and home menu is shown, still holding vol buttons
5. Nothing more is happening

I tried that on my regular switch but the calibration didnt start there, Im doing something wrong.

Ok I understand, if there is anything I can help with on that matter please let me know.

CTCaer, 2021-03-29:
There's no visual confirmation. If it worked it worked. Calibration doesn't help if touch panel's connector is unseated or panel has issues anyway.

Rodhi862, 2021-07-08:
It worked for me to activate the Autoboot from the .ini, because I am having exactly the same mess at the moment on my Switch Lite, thanks for the information provided, so I will be able to continue using the lite.