CTCaer/hekate #592: new switch update

onejman, 2021-04-06:
not working new switch update

impeeza, 2021-04-07:
Is Atsmosphére the one which must to be updated to support the new firmware, and if is necessary Hekate will be updated then.

jaekbrandon, 2021-04-07:
Its giving me the "HOS version is not supported" error when I try to launch the CFW from Hekate.

Adrian-dev-sys, 2021-04-07:
This is not related to atmosphere. Hekate needs to be updated in order to be able to boot sysnand (without CFW) on 12.0.0

peltos, 2021-04-07:
Yeah, i have this problem currently too. See image below when I pressed the CFW (sysMMC) button

[image]

jaekbrandon, 2021-04-07:
Yeah so judging from what the others are saying, I guess we'll just have to wait for a few days or weeks till Hekate is compatible with 12.0.0.

impeeza, 2021-04-07:
Or downgrsde usung a *NAND backup

Masamune3210, 2021-04-07:
Yes, hekate has to be able to parse the firmware to be able to boot it, so it has to be updated as well whenever a major firmware update comes out. You will either have to boot without hekate and burn your fuses, wait for a update, or downgrade using a backup from a previous version on that system

urherenow, 2021-04-07:
@Ron Pelt

CFW (sysmmc)... is still CFW. That not booting is Atmosphere that needs the update. And if your switch isn’t already banned, I would say foolish.

STOCK not booting points to Hekate (which I’m sure is the case as well this time, I’m just trying to clarify things).

mdardir2002, 2021-04-08:
So I do have a RawNand backup using Hekate for my system with 11.0.1 on it. I accidentally updated to 12.0. if I use Hekate to restore my system back to 11.0.1 will that blow any of the fuses because I did a downgrade?

Masamune3210, 2021-04-08:
No, as long as you boot with hekate and have noGC on, your fuses are untouched

mdardir2002, 2021-04-08:
thanks for the fast reply. I'm new to doing this so excuse my ignorance. I can access Hekate but what's noGC on ?? and do you know of a tutorial or a video showing a downgrade using an older Rawnand?

impeeza, 2021-04-08:
Please go to proper channels to get support, this space is for report issues of Hekate. There is a lot of documentation on the net about all the concepts.

I strongly recommend you to understand all concepts before continue. You can get things worst if you do not know what are you doing, saying that, noGC is a configuration of the loader which block the Game Cardtridge use, so no auto update of firmware (with the nefastus fuses burnt) can be maked by games. That is configurated on hekate_ipl.ini or NYX interface.

And, if you restore a backup and never start original firmware without atmosphère the fuses never get burn

CTCaer, 2021-04-09:
Support was added for 12.0.0.
Other than that, you should really wait.

If you managed to update your OFW and stock option doesn't work,
either disable autorcm or compile -dev branch or get the binary from hekate-support channel in retronx discord.

Vordred641, 2021-04-10:
@CTCaer 
Hi, thank you for the update but I was just wondering on approximately how long should we wait for hekate to be compatible with 12.0.0? Many thanks.

urherenow, 2021-04-10:
hekate is a bootloader (bundled with the awesome tool which is nyx), not a CFW. You have to wait for Atmosphere to get updated.

Vordred641, 2021-04-10:
Ah ok thanks, how long(aproximately) would atmosphere with sept take to get updated?

stromnov, 2021-04-10:
> Ah ok thanks, how long(aproximately) would atmosphere with sept take to get updated?

From two or three weeks. Or never.

https://github.com/Atmosphere-NX/Atmosphere/issues/1414

Vordred641, 2021-04-10:
Ah I see. Can @CTCaer confirm this please?

urherenow, 2021-04-10:
1) he has nothing to do with Atmosphere 

2) Github is not a support forum.

Stop bugging the developers and go ask your questions on gbatemp or discord. Except for "when" questions. The answer to that usually devolves into "when your dumb, lazy ass learns how to program and do it yourself"

Vordred641, 2021-04-10:
Thank you very much for your help and I apologise for my ignorance (I'm new to this stuff).

CTCaer, 2021-04-11:
Check releases

closed 2021-04-11 by CTCaer

DarkRaGaming, 2021-04-12:
> hekate is a bootloader (bundled with the awesome tool which is nyx), not a CFW. You have to wait for Atmosphere (with sept) to get updated.

Atmosphere got updated yesterday .

impeeza, 2021-04-12:
> > hekate is a bootloader (bundled with the awesome tool which is nyx), not a CFW. You have to wait for Atmosphere (with sept) to get updated.
> 
> Atmosphere got updated yesterday .

Yes, and hekate too some minutes later please read the otgers post and go to releases