CTCaer/hekate #595: Hekate 5.5.5, HOS 12.0.0 sleep broken

ajbritain4, 2021-04-12:
Hey, having an issue with the newest Hekate on my Switch. It’s a patched Erista model with SX Core chip flashed with spacecraft-nx, with Hekate as the payload.bin

Booting stock through Hekate 5.5.4 and HOS 11.0.1 on the sysMMC, stock boot worked absolutely fine. 

With 5.5.5 and HOS 12.0 sysMMC, the system hangs to a black screen on sleep.

5.5.5 and CFW emuMMC on 11.0.1 works absolutely fine. Thanks for reading and let me know if you need any more info

all4free, 2021-04-12:
Switch lite on SX lite chip. Hecate 5.5.5. Updated emunand to 12.0.0. when loading emunand blank screen.

ajbritain4, 2021-04-13:
> Switch lite on SX lite chip. Atmosphere 0.19.0. Hekate 5.5.5. Updated emunand to 12.0.0. when loading emunand blank screen.
> OFW 12.0.0 loading successfully.

Think that sounds like a different problem, might be better in a separate issue?

closed 2021-04-13 by ajbritain4