CTCaer/hekate #596: Black Screen While Loading Payload for Switch Version 12.0.0

Recurem, 2021-04-12:
Hi,

I am experiencing issues when loading hekate_ctcaer_5.5.5.bin payload it is giving me a black screen only for Switch Version 12.0.0 when loading the payload via the TegraRcmGUI (version 2.6). It is saying that I injected the payload successfully but nothing happens just black screen.

The bootloader folder contains the folder below in my SD Card. There are some other files but it is related already to the latest version of Atmosphere which is Atmosphère 0.19.0

[image]
[image]
[image]

emummc.kipm
libsys_lp0.bso
libsys_minerva.bso
nyx
res.pak

I already ensured I copied the latest files into the bootloader folder in the SD Card.

I can guarantee as well my switch is unpatched given I was able to load also the latest TegraExplorer.bin with no problems and also loading @CTCaer check.bin file provided here it says "RCM Exploited Device/ VBAT: <values>"

Recurem, 2021-04-13:
I can confirm that my switch is unpatched since it smashed the stack with ID code ox7000 byte

I was able to run the hekate_ctcaer_5.5.5.bin if I do not have contents on my SD card folder. Now I am not sure anymore which is the cause.

[image]

Recurem, 2021-04-13:
[image]

urherenow, 2021-04-13:
You didn't show a pic of the sys folder, you haven't established that your card is formatted correctly and tested for errors and being fake (h2testw), or that your sd card socket isn't damaged.

Are you able to see the contents on your card when you launch OFW? Take screenshots and then check your album, while in OFW...

Your error clearly states that it can't find nyx (which should be nyx.bin in your sys folder.

Recurem, 2021-04-13:
@urherenow  Which sys folder?

I can verify my card is formatted correctly. I am using this card on may separate Switch (which is already running homebrew).

Yes, the nyx is not there because I am showing that I am able to load the hekate_ctcaer_5.5.5.bin (which is above screenshot) if I did not put the bootloader files into my SD Card.

It is only having issues when I put the bootloader files into my SD Card then load it using TegraRcmGUI the payload hekate_ctcaer_5.5.5.bin showing as black screen afterwards.

urherenow, 2021-04-13:
bootloader/sys

What is in your hekate_ipl.ini?

Recurem, 2021-04-13:
@urherenow  Here is the configuration for the hekate_ipl.ini. No I am not bypassing nyx, the screenshot I showed earlier loads when I tried to load the hekate_ctcaer_5.5.5.bin without any files in my SDCard that is why it is saying the nyx is probably missing.

When I put all the bootloader files into my SDCard and tried to load the hekate_ctcaer_5.5.5.bin. It only black screens after attempting to load the payload successfully based on the input from Tegra payload loader.

[config]
updater2p=1
{------ Atmosphere ------}
[Atmosphere FSS0 EmuMMC]
fss0=atmosphere/fusee-secondary.bin
kip1=atmosphere/kips/*
emummcforce=1
icon=bootloader/res/emu_boot.bmp
{}
[Atmosphere FSS0 SYS]
fss0=atmosphere/fusee-secondary.bin
kip1=atmosphere/kips/*
emummc_force_disable=1
icon=bootloader/res/sys_cfw_boot.bmp
{}
{-------- Stock ---------}
[Stock SYS]
fss0=atmosphere/fusee-secondary.bin
stock=1
emummc_force_disable=1
icon=bootloader/res/stock_boot.bmp

urherenow, 2021-04-13:
Any reason why you're missing most of the basic config settings? 

[config]
autoboot=0
autoboot_list=0
bootwait=3
backlight=100
autohosoff=0
autonogc=1
updater2p=1
bootprotect=0

Recurem, 2021-04-13:
@urherenow I already tried adding the basic config settings. And loaded hekate_ctcaer_5.5.5.bin still on blackscreen after Tegra saying payload successfully injected.

urherenow, 2021-04-13:
dm me on gbatemp. we should pick this up there because this isn't meant to be a forum. Have you tried launching OFW yet? You can take a screen shot and see it in the album? If so, what is the crc32 of your nyx.bin (check on the card, not within the .zip... I use hashtab for windows, myself...). 

Also, I haven't heard of this being a problem in a long time, but if you look at the properties of hekate_ctcaer_5.5.5_Nyx_1.0.2.zip, does a check box appear near the lower end of the window, giving you the option to "unblock" it? iirc, if that was even a problem on your system, then the crc32 of the file extracted to your card would be incorrect...

Recurem, 2021-04-13:
Seems like the issue is on the libsys_minerva.bso file which causes to blackscreen when loading any payloads. After I tried to rename or remove the file I was already able to load Hekate but it prompts as an error in which the file is missing for Minerva.
[image]

[image]

I am not sure what is causing this. Attaching screenshot for refence

urherenow, 2021-04-13:
I've already told you, github is not a help forum. It's for actual bugs.

The date/time thing is normal for a first run. You need to delete your hekate .zip file then download it again, because something is wrong with the file(s) you're using. CTCaer would probably want you to to post a screenshot of your sd information as nyx shows it, since it appears you've been able to at least boot to the gui...

CTCaer, 2021-04-13:
@Recurem you either have corruption or your sd card reader connector has issues.
Or your cable might have issues.
I would check for these in your case.

If it's corruption, repaste release and make sure you safely eject.
If the dir was corrupted, then you need to copy the contents of your hekate ini, delete bootloader folder, repaste anew and run once. If it's fixed, paste the ini contents back.

Recurem, 2021-04-13:
Hi @CTCaer 

Thank you for attending to my concern. I am not sure anymore this might be a hardware/software specific on my Switch since this is my first time in attempting homebrew on that specific switch unit (this is a V1 unpatched), because I have another separate V1 unpatched Switch here (FW 11.0.1). I was able to load the Hekate without issues with Minerva contents successfully by putting same SDCard. If there was really a SD Card Reader corruption then my another switch should not be able to load the payload but it was able to load without issues. I also noticed when the Minerva module is there it prevents me to load other payload successfully (e.g TegraExplorer or Lockpick) but I am able to load it on my other separate switch as well 

I hope we can agree at some point that it might be a Switch unit specific (it maybe a hardware or software related) since we already know the SD contents with Minerva I used works on a separate switch. If it was not working then I would agree this is really a corruption on the SD Card reader.

Masamune3210, 2021-04-13:
Its very likely to be your sd card reader, as they fail pretty easily and cause all sorts of issues when you do anything with the sd card. Good news is the reader is super easy to replace, it's just plugged in under the back panel and can be easily removed and replaced

Recurem, 2021-04-13:
I just want to iterate If this was a card reader issue then both switches that I have should have black screen when I load the payload using same SD card, I just literally swapped the SD Card to my another switch and load the Hekate payload and it was able to load it successfully without any issues or black screen.

I am not sure anymore what do do this unit I am trying to homebrew. I might have bad luck.

Masamune3210, 2021-04-13:
Why would it be both switches? If the same sd card loads on one switch but not the other, one would assume that the card reader on the switch that didn't load correctly is probably faulty

Recurem, 2021-04-13:
Yes, Now I understand and would agree it might be a card reader on the Switch itself. But one thing for sure I am wondering it is only affecting libsys_minerva.bso preventing any payload/s from being loaded.

I guess I just really have a bad luck with this unit.

CTCaer, 2021-04-13:
@Recurem put back minverva and try this in nyx.ini `newpowersave=0`
Tell me results.

Recurem, 2021-04-13:
Hi Sir @CTCaer where do I modify this newpowersave=0? I cannot seem to find nyx.ini in my bootloader folder.

Sorry I am quite new to this.

CTCaer, 2021-04-13:
It should have been already created if you clicked done on that screen https://github.com/CTCaer/hekate/issues/596#issuecomment-818697468

Just create a new file named `nyx.ini`
With these contents:
```ini
[config]
timeoff=1
newpowersave=0
```

Recurem, 2021-04-13:
I tried setting it up to that configuration mentioned @CTCaer still the same with black scree when I tried to load the hekate_ctcaer_5.5.5.bin payload with the Minerva module.

Do you need more details?

CTCaer, 2021-04-13:
Well that's weird. And more importantly screen artifacts you get when you are finally in Nyx.
I can possibly pinpoint if the issue is your RAM chips or SD card reader **connector** (not the reader itself as others pointed here. That one is difficult to break), or something else.
But you will need to contact me in retronx discord server. Join in and pm me.

Recurem, 2021-04-13:
Hi @CTCaer I already added you on RetroNx as Elcian. Please check and thank you.

CTCaer, 2021-04-13:
Issue is that a RAM chip is faulty. Specifically a ram bank of one of the 2 chips.
User was advised to repair it or try to return it back.

closed 2021-04-13 by CTCaer

urherenow, 2021-04-13:
Yea, sorry the extra help started when I was asleep. I would have mentioned that he "received it with FW 12 on it already". So it's used. Nobody is going to sell an unpatched unit unless it's broken or you agreed to pay way more than retail...

marmar12244, 2023-07-19:
tegrarcm saying successfully  injected payload and nothing popping up on my switch any tips?

Masamune3210, 2023-07-19:
Please don't post on dead issues. Either your switch is patched or you are doing something wrong, likely the first one though. Regardless, it's not an issue with Hekate.

marmar12244, 2023-07-22:
Ok thanks