CTCaer/hekate #600: can't build

costelabr17, 2021-04-22:
no matter what i try always shows me this errors
make: tools/lz/lz77: No such file or directory
make: *** [Makefile:118: loader] Error 127

urherenow, 2021-04-22:
You can't just download a .zip of this repo. You need to clone it recursively to ensure you have all submodules. If you are sure you did a clone xxxxx --recursive, then you may need to double-check what's installed or needs updating from devkitpro (dkp-pacman).

costelabr17, 2021-04-22:
> You can't just download a .zip of this repo. You need to clone it recursively to ensure you have all submodules. If you are sure you did a clone xxxxx --recursive, then you may need to double-check what's installed or needs updating from devkitpro (dkp-pacman).

i  already do this 

make[1]: *** [Makefile:16: lz77] Error 127
make: *** [Makefile:129: tools/lz] Error 2

and now this error

impeeza, 2021-04-23:
You need to have installed the devkitARM toolchain for me was:
`pacman -S devkitARM`
command on my MSYS64 installation

costelabr17, 2021-04-23:
> You need to have installed the devkitARM toolchain for me was:
> `pacman -S devkitARM`
> command on my MSYS64 installation
> 
> and the switch-zlib package, also `pacman -S --needed --noconfirm switch-zlib` command

0 [main] gcc (13588) C:\cygwin64\bin\gcc.exe: *** fatal error - cygheap base mismatch detected - 0x180352408/0x180347408.
This problem is probably due to using incompatible versions of the cygwin DLL.
Search for cygwin1.dll using the Windows Start->Find/Search facility
and delete all but the most recent version.  The most recent version *should*
reside in x:\cygwin\bin, where 'x' is the drive on which you have
installed the cygwin distribution.  Rebooting is also suggested if you
are unable to find another cygwin DLL.
make[1]: *** [Makefile:16: lz77] Error 127
make: *** [Makefile:129: tools/lz] Error 2

urherenow, 2021-04-23:
Eww. Don't recall any tutorial recommending cygwin. Recommended for Windows is Msys2 (with mingw64, if you build Windows apps as well).

This is not a help forum and your problem is not because of a bug. If you can't get help on Discord, make a thread in gbatemp and close this issue with a link to your thread. I can help from there.

CTCaer, 2021-04-23:
What you have fails to compile the tool before running it. So try to run a make without concurrent -j# options.
Other than that dunno. It always works in wsl1/2 and linux.

Anyway, Msys or anything else other than WSL1/2 is not recommended for hekate and thus no support is given.

closed 2021-04-23 by CTCaer