CTCaer/hekate #602: Freezing when trying to backup (USB Tools > BOOT0)

Whitesttax, 2021-04-24:
[image]
Trying to install atmosphere on a 12.0.1 Switch. After placing bootloader with latest hekate on SD and injecting 5.5.5 I go to Tools > USB tools > Boot0 and it won't show up on Windows as a drive so I can't use NxNandManager. Also if I try to remove the USB C cable hekate freezes and I need to hold power to turn it off.

CTCaer, 2021-05-05:
The card is legit and I don't see other errors.
Contact me in retronx discord, I think I know what's the issue.

CTCaer, 2021-05-16:
If you still have that issue contact me or directly use the new option in nyx.ini (currently it's only editable manually and not via ui)

closed 2021-05-16 by CTCaer