CTCaer/hekate #604: Switch can not be detected by Windows

Initializist, 2021-04-30:
I'm on Hekate 5.5.5. When I tried to mount the SD card in the switch by USB tool, the screen showed that the status is started UMS.  However, my laptop played a sound and not recognized the device. When I check the "Devices and Printers" on my laptop, my switch is listed as UMS. 
[image]

impeeza, 2021-04-30:
Yes that's correct, if the problem is you don't see the «drives» you need to open storage manager and assign letter to the SD card volume. Take care to no edit any emummc partition of you have one

Initializist, 2021-04-30:
Thank you! I will try to see if it works.

Initializist, 2021-04-30:
It got fixed, Thank you again!

closed 2021-04-30 by Initializist