CTCaer/hekate #605: v5.5.5 and ofw 12.0.1 keeps crashing/freezing on sleep or boot

mistertoze, 2021-05-01:
I'm having trouble keeping my Switch on after updating to 5.5.5 and OFW 12.0.1.

Sometimes when launching OFW from hekate, the screen just stays dark and I need to hold the power button to get TegraRCM to go green and inject the payload to go back to hekate.

But most times, if I let the console go to sleep, it seems to crash. With the cable connected, TegraRCM goes immediately green when the console goes to sleep. Other times, it freezes on a dark screen and I have to force a reboot.

This wasn't happening with 5.5.4 and 12.0.0.

impeeza, 2021-05-01:
Could be corrupted files on your SD. Install again atmosphère and Hekate seems like the sleep binary is incorrect or damaged

mistertoze, 2021-05-01:
I did a clean install of both before so it didn't occur to me that it could be related to the files.
So I did it once again and it still gave me two dark OFW boot screens but now it seems to have stopped.

And what correct format? I didn't see any template to report an issue here.

closed 2021-05-01 by mistertoze