CTCaer/hekate #606: update for latest tools

urherenow, 2021-05-01:
I updated devkitpro because Atmosphere 0.19.2 uses the updated devkitARM/devkitA64 releases. Hekate still compiles with this, but with a ton of [-Warray-bounds] warnings in nyx, and one in function loader_main.

loader.c: In function 'loader_main':
loader.c:103:9: warning: 'memcpy' offset [0, 131] is out of the bounds [0, 0] [-Warray-bounds]
  103 |         memcpy((u8 *)(IPL_LOAD_ADDR + IPL_PATCHED_RELOC_SZ), &b_cfg, sizeof(boot_cfg_t));
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

CTCaer, 2021-05-05:
I will up a gcc 11 commit when I validate everything's fine.
The warnings can be ignored. There are other things that gcc 11 breaks for now.

CTCaer, 2021-05-12:
Both master and dev branches have fixes for GCC 11 now.

closed 2021-05-12 by CTCaer

impeeza, 2021-05-12:
Thanks a lot @CTCaer