CTCaer/hekate #607: Version 5.5.5 Problems

renegade2k82, 2021-05-02:
So this breaks 1.0 with pegascape.I have the correct files in the pegascape folder on the root of the sd card and the hekate payload named as reboot_payload.bin in the atmosphere folder.Well when i go to reboot into rcm from clicking the nebera icon from pegascape all i get is a flickering black screen.Also the previous version i got working half the time now all it says is success after clicking nebera on pegascape and my switch doesent even reboot at all.I know from the release notes for pegascape it looks for the payload in atmosphere folder unlike before where it would look for it in the nereba folder on the root plus it was working at one point so something must be going on with hekate.

CTCaer, 2021-05-05:
I checked nereba and the issue is that it assumes small payloads when it can use bigger ones.
For now just copy hekate verbatim for release.
Also the one I know uses nereba.bin. Which one you used?

CTCaer, 2021-05-05:
In case someone wants a proper fix instead of copying the compressed hekate payload,
fixes for nereba and caffeine can be found at:
https://github.com/pixel-stuck/nereba/pull/1
https://github.com/liuervehc/caffeine/pull/2

renegade2k82, 2021-05-07:
I am using the one from https://github.com/AtlasNX/nereba/releases/tag/v0.2 and it says it uses the payload from the atmosphere folder thats named reboot_payload.bin

renegade2k82, 2021-05-08:
aLSO WHERE IS THIS hekate verbatim for release i can download?I cant find it.One last thing is this issue going to be fixed?

renegade2k82, 2021-05-09:
Um hello?

Masamune3210, 2021-05-09:
Please don't bump a issue like that, at least wait a few days. People have lives and will get back to you when they can

renegade2k82, 2021-05-10:
Ok I understand that and I have a life also plus I have been waiting so thats fine I'll wait not trying to cause anything here.

urherenow, 2021-05-10:
The fix was already linked to up above, and doesn't seem to be directly related to this repo.

CTCaer, 2021-05-11:
Ask the people that provided the binaries to get you one with the fix posted if you don't know how to compile the fix.
Otherwise, for the 3rd time, just copy paste the hekate binary from the release zip that is compressed.

closed 2021-05-11 by CTCaer

impeeza, 2021-05-11:
A RTFM is always welcome :D  he he he

renegade2k82, 2021-05-11:
there is no compressed binary in the zip only the normal hekate bin.Guess i will have to ask them to compile because i defiantly dont know how to compile anything.

CTCaer, 2021-05-12:
yeah that hekate bin in the release is compressed.
It's around 108KB.
And when it gets unpacked into reboot_payload.bin or update.bin it becomes 135KB.

renegade2k82, 2021-05-12:
Ok because that is the bin I have been using and have been having issues with.i unzip the zip rename the bin and it won't work.

CTCaer, 2021-05-12:
safely eject when you write on your sd card. otherwise you can corrupt.
that binary is 108KB so it should work with nereba.

renegade2k82, 2021-05-12:
Ok.I do make sure i eject the sd card before taking it out so im pretty positive nothing got corrupted.

Masamune3210, 2021-05-13:
Maybe failing sd card or sd card reader needs to be reseated?

renegade2k82, 2021-05-13:
I don't think its my SD card reader.i haven't had any problems and no Coruption issues with other SD cards I have used with it.Also the actual SD card I used in my old switch and it worked fine with no issues but I reformatted it and I'll try a fresh payload and see what happens.

renegade2k82, 2021-05-13:
Ok so i updated my atmosphere to the latest 19.3(deleted the old files and started fresh) and updated the hekate files to the latest 5.5.6(also deleted the old files and started fresh) renamed the new payload to reboot_payload.bin put it in the atmosphere folder replacing the previous version loaded up pegascape clicked on nereba and rebbots to a black screen nothing happens.this has me stumped.I did everything to try to make this work like reformatting,deleting the old files and putting the updated ones on it etc.

renegade2k82, 2021-05-16:
What should I do?

CTCaer, 2021-05-16:
Replace your reboot_payload.bin with this:
[rcm_check.zip](https://github.com/CTCaer/hekate/files/6488878/rcm_check.zip)
and tell me what happens

george-lim, 2025-03-02:
Hey all! I don't mean to revive this thread, but since it was super helpful I thought I would share the compiled NSP for both @CTCaer's payload size increase change and the payload path changed to `sdmc:/atmosphere/reboot_payload.bin` (matching AtlasNX's nereba v0.2 changes)

[Release](https://github.com/george-lim/nereba/releases/tag/v0.3)

george-lim, 2025-03-03:
Hey all! Apologies for reviving this incredibly helpful issue, just thought I'd share an updated build of Nereba with both CTCaer's payload size increase change and AtlasNX's v0.2 payload path change. This is mainly useful for those that aren't devs or don't want to deal with recompiling Nereba from source. It looks like the Nereba repo is inactive so I doubt CTCaer's PR is going to get merged.

[Release](https://github.com/george-lim/nereba/releases/tag/v0.3)