CTCaer/hekate #608: hekate_ipl.ini does not update properly ?

T1nd4, 2021-05-06:
hekate_ipl.ini does not update with emuMMC boot options after created emuMMC
- took a new SD
- partition SD card as usual (full)
- goto emuMMC/create emuMMC
When finished "hekate_ipl.ini" is not updated with emu entries in launch section.

I'm 10.0.4 original firmware, using atmosphere 19.2 hekate 5.5.5 pack.
When I use sd card with atmo 18.01, hekate_ipl.ini updated well after created emuMMC.

I fixed the my 19.2 installation with adding next lines in my hekate_ipl.ini
Then all works fine.

[CFW - emuMMC]
fss0=atmosphere/fusee-secondary.bin
kip1patch=nosigchk
emummcforce=1
atmosphere=1
icon=bootloader/res/icon_payload.bmp

[Stock - sysMMC]
fss0=atmosphere/fusee-secondary.bin
emummc_force_disable=1
stock=1
icon=bootloader/res/icon_switch.bmp

CTCaer, 2021-05-11:
Nothing adds sections in hekate_ipl.ini.
User is supposed to add them by studying the readme first or get a ready made.

closed 2021-05-11 by CTCaer

impeeza, 2021-05-11:
Tonight is the RTFM night ^_^   ^_^    ^_^

T1nd4, 2021-05-11:
Huhu...
I never had to setup manually my ini. I thought it was done directly by hekate. That’s probably the problem with all the all in one builds.
Thanks