CTCaer/hekate #609: Touch issue

tjinn, 2021-05-10:
Touch Panel
Model: Unkown Contact me!
ID: 6D139629
FTB ver: 137C
FW ver: 3B5E
Changes on every boot of Hekate.
Happened a while ago but idk what to do now. 
Calibration just doesn't work.

Masamune3210, 2021-05-11:
Have you ever had your touch screen replaced? Have you boot into anything other than HOS?

tjinn, 2021-05-11:
No.
I've only booted into HOS and atmosphere.

CTCaer, 2021-05-11:
That normally happens if you never booted HOS and your touch chip does not have the nintendo customized configuration and thus missing the command that gets the touch panel model info.
Also make sure that hekate and nyx (bootloader folder files) are the latest.

I would normally suggest you to boot HOS once but you said you already booted, so dunno why it didn't update the touch firmware.

tjinn, 2021-05-11:
Is it possible to get this information from a nand backup or something?
This happened a while ago but most of the things i was playing didnt use
the feature.

tjinn, 2021-05-11:
Is there anything I can do to attempt to fix it software wise?

CTCaer, 2021-05-12:
So basically your touch broke.
If that's the case, you can do nothing in hekate to fix it.
Probably your touch panel or GC module flex cable got unseated?

Anyway that's a hw issue that can't be fixed via software.

closed 2021-05-12 by CTCaer