CTCaer/hekate #611: new fw in the wild (12.0.2)

loopoutgh, 2021-05-12:
New HOS fw, 12.0.2. Breaking hekate latest version's capability to boot clean sysnand with ofw through fusee secondary?

impeeza, 2021-05-12:
What is the issue? Did you find a bug?

loopoutgh, 2021-05-12:
Not yet, just giving a heads up on the update since probably will need some update on hekate's part to boot the newer fw as other times before.

CTCaer, 2021-05-12:
12.0.2 was already pushed.
If you want it now, you can compile master or dev branch here (both were updated).

ctotheharrison, 2021-05-12:
Accidently updated the firmware to 12.0.2. 

Now getting a Fatal Error Package1. any ideas on if ive royally screwed up?

impeeza, 2021-05-12:
This space is for reporting bugs and issues.  For supporting go to the propper channels.  If you read this issue and the release oage Hekate already support the new firmware. 

Please **read** the other components documentation about support on 12.0.2 firmware

CTCaer, 2021-05-12:
Update hekate. I already posted a 12.0.2 compatible release.

https://github.com/CTCaer/hekate/releases/latest

closed 2021-05-12 by CTCaer