CTCaer/hekate #612: black screen when rebooting to payload

MMJv3, 2021-05-12:
i updated to the latest version of hekate with atmosphere and injected at least 1 time with tegra but whenever I reboot to payload it breaks and just shows me a black screen. i can still interact with the buttons but cant see. how can I fix this?

CTCaer, 2021-05-12:
What do you mean you can interact with buttons but can't see?
r2p works fine here, both in 12.0.2 and 4.0.1 w/ 0.19.2 atmo and hekate 5.5.6. Tried it via the power menu.

You can try to replace reboot_payload.bin directly in case it got corrupted if you want.

TattoGM, 2021-05-14:
to black screen solution , download files to SD in this link:

https://www.sdsetup.com/console?switch

Masamune3210, 2021-05-14:
Downloading a prepack would change literally nothing, please don't recommend people do that

TattoGM, 2021-05-14:
> Downloading a prepack would change literally nothing, please don't recommend people do that

It also gave me a black screen, when checking the bootloader folder there is no hekate_ipl.ini file, so when downloading the package, in this if you go to the mentioned file.

update: when installing a copy of XXX game, it tells me that it is not possible to open the program

CTCaer, 2021-05-16:
From lack of reply, seems that OP fixed the issue.
Closing.

closed 2021-05-16 by CTCaer

MMJv3, 2021-05-24:
sort of opened this but forgot. however, the issue isn't fixed and I apologize for the lack of reply. To put it into perspective whenever I use reboot to the payload to enter Hekate I am greeted with a black screen. I can interact with all of the buttons but I am unable to actually see what I'm pressing. the screen is just black but turned on. however, whenever I am on emummc or system, and I restart the console to boot to Hekate, I can see everything fine.

I've tried updating, changing reboot to the payload to the Hekate bin file, and reinstalling my atmosphere folder.

I am on 5.5.6 HOS 12.0.2

I'm not very knowledgeable when it comes to this so go slowly for me. And again I apologize for forgetting to reply to this. I found it again and decided to add some clarification to my issue

CTCaer, 2021-05-26:
@MMJv3 
>I can interact with all of the buttons

That's actually very interesting.
Assuming you do r2p properly.

Find me in retronx discord so I can validate some things.
You will also need to define "whenever I use reboot to the payload to enter Hekate" vs "whenever I am on emummc or system".

MMJv3, 2021-05-27:
@CTCaer would you mind linking it for me? i cant seem to find it

MMJv3, 2021-05-27:
nevermind, i've sent you a dm on discord

Majesty
#9519

Xxnoctisx, 2022-01-27:
> 

Hello, I have the same problem. Is there a solution? Greeting